guysoft/FullPageOS #184: How to disable shortcuts?

Manius71, 2017-11-11:
Thank you for your work. There is only one thing I do not know how to solve. I would like to disable the ability to use keyboard shortcuts built into Chromium. How to do it?

guysoft, 2017-11-13:
Er, I just googled and pasted the first answer from askubuntu, that solves it: https://askubuntu.com/questions/162700/how-can-one-disable-shortcuts-of-chrome

closed 2017-11-13 by guysoft

Manius71, 2017-11-14:
Yes. I used this before you wrote, but this doesn't solve the problem. It deprives me of other important shortcuts: Ctrl+Alt+F(1-7) or Ctrl+Alt+Del.

Manius71, 2017-11-14:
Let's be serious. Probably everyone starts looking for a solution on Google.

Well... I used this before you wrote, but this doesn't solve the problem. I found the best solution here [Ctrl key lock] (https://stackoverflow.com/questions/5637530/disable-shortcuts-in-google-chrome), but it deprives me of other important shortcuts not related to Chromium: Ctrl+Alt+F(1-7) or Ctrl+Alt+Del.

Thanks for the answer. Although unfortunately not helpful.

guysoft, 2017-11-14:
Well, sorry, but I can't modify chromium itself. You might be able to edit and recompile it, but that is a challenge which would require both building and modifying the source code.

Manius71, 2017-11-14:
No problem. I just thought someone had a similar problem and found a better solution. Who asks... not stray.