guysoft/FullPageOS #185: 404 loading errors

cloudybear, 2017-11-13:
#### What were you doing?

Adding sites via fullpagedashboard remotely with the web UI (i.e. www.cnn.com , www.google.com)

#### What did you expect to happen?

websites too load

#### What happened instead?

sites on the fullpagedashboard loading "404 - Not found"

#### Was there an error message displayed? What did it say?

404 not found.

I know wifi is working since I added a website to the fullpageos.txt file and commented out the dashboard URL and any URL in that file works fine.

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

2017-10-29_2017-09-07-fullpageos-stretch-lite-0.9.0

guysoft, 2017-12-06:
If you press ctrl+T and put the websites in the address bar - do they load?

guysoft, 2018-02-16:
closing if no reply

cloudybear, 2018-02-16:
Sorry you can close this out. I haven't had time to look into this. I'll
reopen when I ever get back to this.

guysoft, 2018-03-07:
Closing until you repoen

closed 2018-03-07 by guysoft