guysoft/FullPageOS #186: How to update fullpageOS via SSH

benjacksonaus, 2017-11-16:
Hi,
Just wondering how to update the fullpageOS via SSH without losing the configured webpage.

guysoft, 2018-03-07:
Sorry just saw now.
If you mean just the single page without FullPageDashoard, then just edit ``/boot/fullpageos.txt``[ as explained in the README](https://github.com/guysoft/FullPageOS#features). FullPageDashboard is unmaintained, you could probably get it to do that via the mymsql database.

closed 2018-03-07 by guysoft

Murugesh-Hobbyist, 2024-06-01:
Hey Pal! good work!
I couldn't locate or edit the /boot/fullpageos.txt via SSH.
I hope thats what @benjacksonaus may pointed.

mm@raspberrypi:/home $ cd mm
mm@raspberrypi:~ $ ls -lrt
total 0
lrwxrwxrwx 1 root root 23 Mar 31 08:38 scripts -> /opt/custompios/scripts
mm@raspberrypi:~ $ cd scripts
mm@raspberrypi:~/scripts $ ls -lrt
total 48
-rw-r--r-- 1 mm   mm   2488 May 10  2020 rotate.sh
-rwxr-xr-x 1 mm   mm    981 Feb 26 19:08 enable_gpu
-rwxr-xr-x 1 mm   mm    203 Mar 23 10:44 update_lightdm_conf
-rwxr-xr-x 1 root root 1095 Mar 23 14:53 start_chromium_browser
-rwxr-xr-x 1 root root  130 Mar 23 14:53 setX11vncPass
-rw-r--r-- 1 root root  442 Mar 23 14:53 safe_refresh
-rwxr-xr-x 1 root root   63 Mar 23 14:53 reload_fullpageos_txt
-rwxr-xr-x 1 root root  149 Mar 23 14:53 refresh
-rwxr-xr-x 1 root root  174 Mar 23 14:53 get_url
-rwxr-xr-x 1 root root  165 Mar 23 14:53 fullscreen
-rwxr-xr-x 1 root root  529 Mar 31 09:40 run_onepageos
-rwxr-xr-x 1 mm   mm    560 Mar 31 09:44 start_gui
mm@raspberrypi:~/scripts $ pwd
/home/mm/scripts