guysoft/FullPageOS #187: Use with VPN

xvilo, 2017-11-17:
Hello,

Is there anyone succeeded with using this with an VPN connection?
I would like to use this to access an internal (vpn only accisble) website, but i can't seem to get it started on boot! 

I hope somebody could help me out. 

When using `$ openvpn --connect configFileOfVpn.ovpn` in another tty and then switching back and reloading it all loads fine. 

I have added the command to /etc/rc.local as:
`$ openvpn --connect /home/pi/configFileOfVpn.ovpn &` but that didn't work. I hope somebody can help me out with this!

pide2000, 2017-11-22:
I use several pi's with fullpageos over vpn. Best way is to use systemd instead of rc.local.
If you have openvpn installed use these steps.
symlink your ovpn config to /etc/openvpn with the ending .config
`ln -s  configFileOfVpn.ovpn /etc/openvpn/whatever.conf`
Start openvpn with configuration file whatever
`sudo systemctl start openvpn@whatever`
To automatic start up with vpn enable it
`sudo systemctl enable openvpn@whatever`
With 
`sudo systemctl status openvpn@whatever`
you can check the status of the vpn connection.

To automagically get nameserver (if needed) use these lines at the very end of your ovpn config file
`
# configure pushed DNS
script-security 2
up /etc/openvpn/update-resolv-conf
down /etc/openvpn/update-resolv-conf
`

pide2000, 2017-12-15:
@xvilo is the problem solved? So maybe we can close this issue.

guysoft, 2018-01-02:
Closing if no reply

closed 2018-01-02 by xvilo