guysoft/FullPageOS #190: I can't modify the resolution through boot/config.txt

Bolandish, 2017-11-23:
#### What were you doing?
I'm trying to set the resolution correcty

#### What did you expect to happen?
The resolution should change

#### What happened instead?
Nothing happens

#### Version of FullPageOS?
8.1

#### Screenshot(s) showing the problem:

[image]

guysoft, 2017-11-26:
Can you please provide your ``boot/config.txt``?

guysoft, 2017-11-29:
Closing of no reply

Bolandish, 2017-11-29:
@guysoft Sorry i was offline.. Let me provide it tomorrow.

Fowesed, 2017-12-08:
So I obviously posted a duplicate. Same problem here, my config.txt is attached.
[config.txt](https://github.com/guysoft/FullPageOS/files/1542209/config.txt)

Bolandish, 2017-12-11:
Ah, thanks @Fowesed :)

Fowesed, 2017-12-12:
Update: I added hdmi_force_mode=1 and changed hdmi_group=2 & hdmi_mode=82. No change, problem still exists.

407pilot, 2018-02-15:
What is the status on this? 

I'm having the same problem with 0.8.1. Top, Bottom, Left and Right sides are cut off. Overscan and forced console size in Config.txt don't have any effect on the screen size. I even forced Group 2, Mode 0x27, as the TV manual says the TV supports this resolution. Nothing I have tried has made any difference.

This isn't a zoom issue, as text is also cut off in the terminal.

guysoft, 2018-02-15:
@407pilot DId you try changing HDMI settings? Do you get the same problem with normal Raspbian?

guysoft, 2018-03-07:
Closing if no reply.  Unless anyone has anything else to add.

HugoStiglitz159, 2018-07-25:
Bumping this.

Same issue as above posts, no matter what I do with /boot/config.txt the resolution does not change.

Using a Raspi Model 3 B
Native Screen Resolution is 1920x1200
during the boot it shows a 1920x**1080**, but when the page displays its at 1650x1050

I have tried another OS in the same PI and its find, posts at 1920x1200 and displays pages in the same resolution

I have tried the following in the config.txt

disable_overscan
hdmi_force_mode=1
hdmi_group=1
hdmi_mode=16

as well as mode / group (2/69)

none of these modifications change the resolution at any point.

407pilot, 2020-05-28:
> @407pilot DId you try changing HDMI settings? Do you get the same problem with normal Raspbian?

@guysoft sorry for the long delay in replying. After hammering at this problem for a long time, I gave up. Since then I've upgraded to 0.11.0, but still have the same problem. I've tried changing HDMI settings, forcing it to the correct mode for the TV, and changing overscan settings. All of those changes had no effect, positive or negative. Nothing seems to make it better or worse.

At bootup, the screen initially sizes correctly, but at some point during boot (it passes too quickly for me to see) the display settings change.

guysoft, 2020-06-09:
Closing if anyone doesn't have anything else to add.