guysoft/FullPageOS #194: Performance

Imburr, 2018-01-07:
#### What were you doing?

Loading a single LAN web page

#### What did you expect to happen?

the page loads, then it goes blank white. I have a process to reboot chromium every 15m, and most times it brings up the web page again for a bit, until it turns white.

Sometimes when I load the web pages they 404 even though they become available if I reload immediately, and the pages are available on other devices.

#### Was there an error message displayed? What did it say?

High Load, I checked 15m after boot and load averages were 14, 13, 6

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1

#### Screenshot(s) showing the problem:

I have ran htop and noticed that swap is pinged red at 100% for an extended period of time. Or I will see a random CPU core pegged at 100% like so: http://prntscr.com/hx2zhw

Any ideas? Should I try a new/better SD card, or just reinstall?

Imburr, 2018-01-07:
I reinstalled from the latest nightly. Update and upgrate. Set my URL, and reboot. It now loads the page, then flashes white, then flashes 404, then flashes back to the site... all in random timing. I hit up htop and this is what I see: http://prntscr.com/hx3n3a

The page I am loading is a HTTP camera stream page presented by a BlueIris web server.

Here is the web page code in full:

`<html xmlns="http://www.w3.org/1999/xhtml"><head>
	<meta http-equiv="Content-Type" content="text/html;charset=utf-8">
	<meta http-equiv="cache-control" content="max-age=0">
	<meta http-equiv="cache-control" content="no-cache">
	<meta http-equiv="expires" content="0">
	<meta http-equiv="expires" content="Tue, 01 Jan 1980 1:00:00 GMT">
	<meta http-equiv="pragma" content="no-cache">
	<meta name="viewport" content="width=800, initial-scale=0.1">
	<title>Cameras</title>
	<style type="text/css">
		body
		{
			background-color: black;
			margin: 0;
			color: Gray;
		}
		table
		{
			border-collapse: collapse;
		}
		td
		{
			padding: 0;
			text-align: center;
		}
		#popupFrame
		{
			position: fixed;
			background-color: #BB0000;
			border: 1px Solid White;
			border-radius: 5px;
			padding: 5px;
			top: 15px;
			width: 300px;
			left: 20px;
		}
		#popupMessage
		{
			color: White;
			margin-bottom: 5px;
		}
		#popupMessage a
		{
			color: #DDDDFF;
		}
	</style>
	<script src="legacy/jquery-1.11.1.min.js" type="text/javascript"></script>
	<script src="legacy/allcams.js" type="text/javascript"></script>
	<script type="text/javascript">
		var camUrl = "//192.168.2.206:81/image/";
		var refreshTime = 25;
	</script>
</head>
<body>
	<select id="CamList" style="display:none;">
			<option selected="" value="index">+All cameras</option>
<option value="@index">+All cameras cycle</option>
<option value="Frog">Frog</option>
<option value="londyn">Kids</option>
<option value="autumn">Autumn</option>

	</select>
	<div id="rootDiv">
	[image]</div>

</body></html>`

guysoft, 2018-01-07:
How does the page load on normal Raspbian Desktop?
I can't debug a website that I don't have access to.

bagaskarawg, 2018-01-14:
Hi @guysoft, thanks for developing fullpageos 👍 

I am having the same problem about the performance, when I try to load this web page https://www.mapplic.com/plugin, especially when entering fullscreen mode, I feel that it's too laggy when clicking on the maps, I am using Raspberry Pi 3 Model B with 1 GB of RAM. When using Raspbian, there's no laggy problem with the web page. I am planning to use the Mapplic plugin for my upcoming project.

Thank you :)

benyanke, 2018-01-18:
Maybe I'm misreading your issue, but according to your HTOP screenshot, HTOP itself is the process pegging a single core....