guysoft/FullPageOS #195: No VNC or FullPageDashboard webpage access

cult-hero13, 2018-01-12:
#### What were you doing?

After boot/reboot, remote VNC and Dashboard page work as expected.  Later (I'm sorry, but I don't know how much time passes), one or both stop working.  By this I mean remote VNC access attempts result in messages such as "Connection failed to ...", and remote Dashboard page access attempts result in the page loading with the "Default" tab and tab bar showing, but the page simply displays a messages such as "Unable to connect".

#### What did you expect to happen?

I expected to be able to remotely access a FullPageOS Raspberry Pi via VNC and FullPageDashboard web page.

#### What happened instead?

Randomly, but surely, remote VNC and FullPageDashboard webpage access stops working.

#### Was there an error message displayed? What did it say?

VNC access attempts result in "Connection failed to ..."
FullPageDashboard webpage access attempts result in "Unable to connect"

#### Version of FullPageOS?

I have 3 installations of FullPageOS, on 3 different Raspberry Pis.  None of them them have the file /etc/fullpageos_version, but the installations were done no more than 1 month ago.

#### Related Question

Is there a way to restart either VNC or FullPageDashboard without rebooting the Raspberry Pi?

guysoft, 2018-01-12:
1. I need to know what version of FullPageOS are you running.
2. Are you using a good power supply with 2A and a GOOD usb cable?
3. You can restart the lighttpd daemon that runs FullPageDashboard and thet lightdm for vnc, but it will restart the browser.

guysoft, 2018-02-16:
Closing if no reply

closed 2018-03-07 by guysoft