guysoft/FullPageOS #196: Unable to connect ot WiFi

Javelin69, 2018-01-13:
#### What were you doing?
Setting up FullPageOS over Wifi to display a dashboard.  Modified fullpageos-network.txt as per instructions.

#### What did you expect to happen?
Was hoping the connection would be straright forward.

#### What happened instead?
Unable to establish an Internet connection over WiFi

#### Was there an error message displayed? What did it say?
There is no Internet connection
ERR_INTERNET_DISCONNECTED

#### Version of FullPageOS?
Have tried these 2 versions
2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1
2017-10-29_2017-09-07-fullpageos-stretch-lite-0.9.0

I live in Australia and my Raspberry Pi3 Model B is fully updated and upgraded as of 13/01/2018.  
I have followed instructions and suggestions from forums and cannot connect to the internet.

I get no errors while booting.  

All changes made in plain text.

No smart quotes.

Tried 2 different wireless networks.

I am new to Linux so it is possible I have missed something rudimentary.

Your help will be appreciated.

I have tried the following 5 configurations.

-------------------1-------------------------
fullpageos-network.txt - MODIFIED
fullpageos-wpa-supplicant.txt - DEFAULT

## WPA/WPA2 secured
iface wlan0-fullpageos inet manual
    wpa-ssid "MYSSID"
    wpa-psk "MYPASSWORD"

## WEP secured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-key "put password here"

## Open/unsecured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-mode managed

#   address:   192.168.250.10
#   netmask:   255.255.255.0
#   broadcast: 192.168.250.255

#auto eth0:1
#iface eth0:1 inet static
#  address 192.168.250.1
#  netmask 255.255.255.0
#  broadcast 192.168.250.255

-----------------2------------------
fullpageos-network.txt - DEFAULT
fullpageos-wpa-supplicant.txt - MODIFIED (Left Country as GB)

country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

# This is only used to configure multiple wifi networks or other advanced wifi features.
# take a look into fullpageos-network.txt instead if you only need basic wifi configuration
# 'man -s 5 wpa_supplicant.conf' for advanced options
network={
  ssid="MYSSID"
  psk="MYPASSWORD"
#}

-----------------3------------------
fullpageos-network.txt - DEFAULT
fullpageos-wpa-supplicant.txt - MODIFIED (Changed Country to AU)

country=AU
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

# This is only used to configure multiple wifi networks or other advanced wifi features.
# take a look into fullpageos-network.txt instead if you only need basic wifi configuration
# 'man -s 5 wpa_supplicant.conf' for advanced options
network={
  ssid="MYSSID"
  psk="MYPASSWORD"
#}

-------------------4-------------------------
fullpageos-network.txt - MODIFIED
fullpageos-wpa-supplicant.txt - MODIFIED (Left Country as GB)

## WPA/WPA2 secured
iface wlan0-fullpageos inet manual
    wpa-ssid "MYSSID"
    wpa-psk "MYPASSWORD"

## WEP secured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-key "put password here"

## Open/unsecured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-mode managed

#   address:   192.168.250.10
#   netmask:   255.255.255.0
#   broadcast: 192.168.250.255

#auto eth0:1
#iface eth0:1 inet static
#  address 192.168.250.1
#  netmask 255.255.255.0
#  broadcast 192.168.250.255

********************************
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

# This is only used to configure multiple wifi networks or other advanced wifi features.
# take a look into fullpageos-network.txt instead if you only need basic wifi configuration
# 'man -s 5 wpa_supplicant.conf' for advanced options
network={
  ssid="MYSSID"
  psk="MYPASSWORD"
#}

-------------------5-------------------------
fullpageos-network.txt - MODIFIED
fullpageos-wpa-supplicant.txt - MODIFIED (Changed Country to AU)

## WPA/WPA2 secured
iface wlan0-fullpageos inet manual
    wpa-ssid "MYSSID"
    wpa-psk "MYPASSWORD"

## WEP secured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-key "put password here"

## Open/unsecured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-mode managed

#   address:   192.168.250.10
#   netmask:   255.255.255.0
#   broadcast: 192.168.250.255

#auto eth0:1
#iface eth0:1 inet static
#  address 192.168.250.1
#  netmask 255.255.255.0
#  broadcast 192.168.250.255

********************************
country=AU
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

# This is only used to configure multiple wifi networks or other advanced wifi features.
# take a look into fullpageos-network.txt instead if you only need basic wifi configuration
# 'man -s 5 wpa_supplicant.conf' for advanced options
network={
  ssid="MYSSID"
  psk="MYPASSWORD"
#}

Thanks

Geoff

Javelin69, 2018-01-13:
Meant to mention that I have no issues connecting to either network with Out of Box Raspbian.  Using DHCP.

Regards

Geoff

benyanke, 2018-01-18:
You seem to have missed uncommenting the #

```
update_config=1

network={
ssid="MYSSID"
psk="MYPASSWORD"
#}
```
should be:
```
update_config=1

network={
ssid="MYSSID"
psk="MYPASSWORD"
}
```

Javelin69, 2018-02-03:
Thanks heaps works a treat.  I knew it would be something stupid, I just couldn't see it.

samscoms, 2018-02-04:
It works for me too.
I had to set wifi in both files.

guysoft, 2018-02-05:
Closing since OP solved issue

closed 2018-02-05 by guysoft

jyy1082, 2018-03-11:
@Javelin69 how about your first configuration?
-------------------1-------------------------
fullpageos-network.txt - MODIFIED
fullpageos-wpa-supplicant.txt - DEFAULT

WPA/WPA2 secured
iface wlan0-fullpageos inet manual
wpa-ssid "MYSSID"
wpa-psk "MYPASSWORD"

I also can not connect wifi by modifying fullpageos-network.txt