guysoft/FullPageOS #198: Webm stops playing.

donisg, 2018-01-22:
I have internal web page with embedded webm video file on it.
After webm file reaches its end it does not star playing from the beginning. It simply stops at the last frame. At that moment RPi is accessible with ssh, I can log in.

No error messages are displayed at GUI.

It works with no issue with 2016-05-10-fullpageos-jessie-lite-0.6.0.zip.
Problem appears with 2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1.zip and 2017-10-29_2017-09-07-fullpageos-stretch-lite-0.9.0.zip.

guysoft, 2018-01-22:
Can you try to disable GPU acceleration?
Its enabled by default on versions after 0.7.0.

donisg, 2018-01-22:
Do You mean raspi-config>8 Advanced Options>A7 GL Driver ?
I have tried Full KMS, Fake KMS, Legacy.

guysoft, 2018-01-22:
0.9.0 is the "nightly" build.
I can also try building a new one and see if the new raspbian from December might help.
But it might be a chromium bug or. Behaviour change too. Have youbtried it on chrome on the desktop?

donisg, 2018-01-23:
I have tried to install 2017-11-29-raspbian-stretch.img ( 2017-11-29, raspberrypi.org) and use chrome on the desktop. It worked with no problem.

donisg, 2018-01-25:
Today I have tried 2018-01-24_2017-11-29-fullpageos-stretch-lite-0.9.0.zip, but nothing new - video does not loop.

guysoft, 2018-01-30:
[image]

Had a few minutes to paste it in to my pi. 
Chromium has an option to loop videos if you right-click on them, you need to enable it for a video to loop. Closing since this is not a FullPageOS issue.
You can also use a loop video tag in an html page. Its better to set that up than relaying on chrome's video player and its local settings. For more in formation on that see: https://www.w3schools.com/tags/att_video_loop.asp

closed 2018-01-30 by guysoft