guysoft/FullPageOS #199: Chromium auto connect when internet established

facetime88, 2018-01-23:
I have setup fullpageos for a kiosk which will load an internet webpage and need to run 24/7 without need human intervention. However, sometime the internet may get down, and the webpage itself may have some javascript that will reload itself, resulting an internet connection error page.

This internet connection error page will stay there forever until I or someone restart the board manually.
Is there any plugin for Chromium, or fullpageos configuration, so whenever an internet connection back established, the page will get reloaded?

Thank you.

guysoft, 2018-01-24:
You can do the suggestion in this comment:
https://github.com/guysoft/FullPageOS/issues/149#issuecomment-359986758

guysoft, 2018-02-16:
Closing if no reply

closed 2018-03-07 by guysoft