guysoft/FullPageOS #200: After password change - VNC now works with both old and new "pi" user password

Thomas-ZA, 2018-01-31:
#### What were you doing?
After altering default password for the pi user - VNC allows connects both with the default password - as well as with the new password

#### What did you expect to happen?
I expect access denied using the old password

#### What happened instead?
VNC access granted with both passwords

#### Was there an error message displayed? What did it say?
No error

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]
cat: /etc/fullpageos_version: No such file or directory 
(This was loaded from a fullpageos image about a month ago- no idea why the version file is not there...)

#### Screenshot(s) showing the problem:

guysoft, 2018-01-31:
You can try deleting the ``/home/pi/.vnc/passwd``? That's where the password is stored.

Thomas-ZA, 2018-01-31:
I managed to find that "x11vnc -storepasswd" will allow you to reset the password. For us there are a number of displays, where the passwords are incremented. With the length passwords we used, it seems the x11vnc app only hashes the 1st few characters of the password - so regardess of the passwords being different - it would still allow VNC access.

I am not yet sure what the maximum password length is - but have found that SIGNIFICANTLY shortening the passwords, allows VNC to behave as expected.

Its frustrating since this is way less secure - but it at least lets the team differentiate between devices a bit easier..

Thanks for your speedy response though!

guysoft, 2018-01-31:
Out of curiosity - how many of them do you have running?

Thomas-ZA, 2018-01-31:
only 6 small screens - forming a dashboard area with alerts

guysoft, 2018-01-31:
6 screens and pis is awesome!
I barely have one.
Closing since this is solved. 

I really need a "post your setup section" somewhere. Curious how it looks now.

closed 2018-01-31 by guysoft

Norfeldt, 2022-04-30:
Thanks! Found the `x11vnc -storepasswd` helpful in changing the password from the default `raspberry` (even though I have changed the pi user password).

guysoft, 2022-05-02:
Cool this is expected behavior