guysoft/FullPageOS #201: 4k display (3840x2160) scaled down to 1080p

wjdp, 2018-02-05:
#### What were you doing?

Trying to get fullpage os to output 4k on 4k screen. The following was added to the config.txt file

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 3840 2160 24
max_framebuffer_width=3840
max_framebuffer_height=2160
hdmi_pixel_freq_limit=400000000
```

https://www.raspberrypi.org/forums/viewtopic.php?p=562907&sid=7eb39d2f1db986aea5dc07eb3a6e7a8e#p562907

On 'normal' raspbian this gives us a 4k desktop.

#### What did you expect to happen?

4k goodness

#### What happened instead?

The startup rainbow and framebuffer are shown in 4k (I can see the crisp edges) but once chrome starts the display drops down to 1080p. Checking via JS reports that size and when pressing F1 to open chrome help in a new 'window' all elements (inc toolbar) is fuzzy indicating screen resolution is 1080p now.

#### Was there an error message displayed? What did it say?

No

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

/etc/fullpageos_version does not exist

2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1.zip is the version I flashed

guysoft, 2018-02-06:
Did you try disabling the GPU acceleration?
I really have nothing special to say beyond that. I have no 4k screen or device to test with.

guysoft, 2018-02-16:
closing if no reply

wjdp, 2018-02-17:
Whoops sorry @guysoft 

I've not had a chance yet, how would I go about disabling GPU acceleration.

I'm using fullpage OS to run a few custom TV dashboards at work and am quite interested in getting 4k running, am happy to give quite a few thing a go if you can provide any pointers.

guysoft, 2018-02-17:
I actually posted here that it was second result in google at the time:
https://github.com/guysoft/FullPageOS/issues/137#issuecomment-346509534
 https://docs.eltechs.com/install-and-configure-exagear-desktop/hardware-graphics-acceleration-on-rpi

guysoft, 2018-03-07:
closing if no reply

wjdp, 2018-03-07:
Thanks for the pointer, just tried.

Can confirm that setting (7 Adv Options, A6 GL Driver, G3 Legacy) along with the settings in my original post gets me a 4k display.

guysoft, 2018-03-07:
Yay!

closed 2018-03-07 by guysoft