guysoft/FullPageOS #202: SSID with a space

FatManLovesPi, 2018-02-05:
I cannot get this connected to an SSID with a space in the name. I have tried quotes and apostrophe around the SSID name but no luck. 

Any ideas on what I need to do to get this to work?

guysoft, 2018-02-05:
Please paste your configuration file

FatManLovesPi, 2018-02-05:
The SSID is "Company WiFi" this is not it but close. It is two works with a one space.

### WIFI CONFIGURATION ######################################################
#The three segments below should cover you in most cases if you run
# a wifi network that uses either WPA/WPA2 or WEP encryption.
#
# Just uncomment the lines prefixed with a single # of the configuration
# that matches your wifi setup and fill in SSID and passphrase.
#
# WPA/WPA2 secured
iface wlan0-fullpageos inet manual
    wpa-ssid Company WiFi
    wpa-psk 112233445566
## WEP secured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-key "put password here"
## Open/unsecured
#iface wlan0-fullpageos inet manual
#    wireless-essid "put SSID here"
#    wireless-mode managed

guysoft, 2018-02-06:
Did you try using the octopi-wpa-supplicant.txt file? Its the recommended one. Revert your changes in octopi-network.txt and use octopi-wpa-supplicant.txt

guysoft, 2018-02-16:
closing if no reply.

marians, 2018-02-18:
@FatManLovesPi Have you tried setting the SSID in quotes?

```
iface wlan0-fullpageos inet manual
  wpa-ssid "Company WiFi"
  wpa-psk 112233445566
```

guysoft, 2018-03-07:
No reply from OP

closed 2018-03-07 by guysoft