guysoft/FullPageOS #204: How to lock-down the kiosk?

mrfixit3000, 2018-02-08:
This package is perfect except it still allows the local user to perform a lot of keyboard shortcuts. I need to know how to disable the TTY switching options, and basically disable all other keyboard shortcuts in chrome and X. It needs to be impossible for a user to do anything outside of chrome, we will manage it entirely using SSH.

mrfixit3000, 2018-02-08:
I found a partial solution just minutes after this (after searching all day of course).

This finally successfully disables the TTY switching. The only thing left is disabling any browser hot-keys:
As root, in the folder /usr/share/X11/xorg.conf.d, create a new file named 50-novtswitch.conf, with its content being:
Section "ServerFlags"
Option "DontVTSwitch" "true"
EndSection

mrfixit3000, 2018-02-08:
Also need to know how to change the default x11vnc password, where is the associated password file located

mrfixit3000, 2018-02-08:
and where to change the URL for chromium to an EXISTING and RUNNING system.

mrfixit3000, 2018-02-08:
In an attempt to do this on my own, I stopped and started the lightdm service, and also changed the website listed in /boot/fullpageos.txt. Now when it boots, I get a blank screen with a mouse cursor :( Please help

guysoft, 2018-02-14:
Thats because fullpageos does not start the browser until it manages to connect to the local web server. You can change this behavior in:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos

Just change that if to 
```
if true; then`
```

guysoft, 2018-02-16:
Regarding lockdown. I see you figured it out. So closing if no reply.

closed 2018-02-19 by mrfixit3000