guysoft/FullPageOS #205: Need to turn on screensaver option

eggmer1, 2018-02-16:
Is there a build/variant where screensaver is optional/not disabled?

Using for actual kiosk pointed at a commercial webpage.  I'm a newbie to linux, but managed to get it do do everything I needed (autoreboot, screen off/on, refresh page, etc) but can't get the screensaver enabled.  Google searches til my eyes bled.  Found many ways to DISABLE, but can't find those in fullpage to reverse.

V.0.8.1

guysoft, 2018-02-16:
Screen saver is disable by default in the image because its used as a dashboard by a great number of users. 
To enable it you need to uncomment this line:
https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/gui/filesystem/home/pi/scripts/start_gui#L2
which is in ``/home/pi/scripts/start_gui``.
There is no variant, but it would not be complicated for you to build one, if you need lots of pis with this.

guysoft, 2018-03-07:
Closing if no reply

bcherup, 2018-12-16:
Uncomment the line ?

guysoft, 2018-12-16:
You need to comment out the line:
https://www.yourdictionary.com/comment-out

closed 2018-12-16 by guysoft