guysoft/FullPageOS #206: Building in Docker (Debian stretch) fails: Module loop not found

marians, 2018-02-16:
#### What were you doing?

I am trying to run the build process on my Mac in Docker, using [`debian:stretch`](https://hub.docker.com/_/debian/) as a base image.

If somebody has that working, I'd appreciate your help.

My Dockerfile so far:

```dockerfile
FROM debian:stretch

WORKDIR /

RUN apt-get update -y
RUN apt-get install -y realpath p7zip-full qemu-user-static git wget kmod

RUN git clone https://github.com/guysoft/CustomPiOS.git
RUN git clone https://github.com/guysoft/FullPageOS.git

WORKDIR /FullPageOS/src/image

RUN wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspbian_lite_latest'

WORKDIR /FullPageOS/src

RUN /CustomPiOS/src/update-custompios-paths

RUN modprobe loop

CMD ["bash", "-x", "./build_dist"]
```

#### What did you expect to happen?

I expected to be able to execute `docker build` to have a Docker image that I could use for building a FullPageOS image.

#### What happened instead?

`docker build` exited unsuccessfully with this error:

```
Step 12/13 : RUN modprobe loop
---> Running in 6090cf643b63
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open
  moddep file '/lib/modules/4.9.60-linuxkit-aufs/modules.dep.bin'
modprobe: FATAL: Module loop not found in directory /lib/modules/4.9.60-linuxkit-aufs
```

#### Version of FullPageOS?

Current default branch, which is `devel` it seems.

guysoft, 2018-02-16:
Not a docker expert, looks like the system does not support loopback devices.
Its failing on the ``modprobe loop`` command, which is a kernel module, that image is missing it.
Quick googeling [returns this](https://serverfault.com/questions/701384/loopback-device-in-a-linux-container). They mention a switch there to make the container support loopback devices.

I never got docker to work (again, not an expert), but there is a vagrant script available and documented.

If you do get a docker method to work would love to add it to CustomPiOS.

marians, 2018-02-16:
Thanks @guysoft for the quick reply! That link helped me out actually. I had to remove the `modprobe` command from the Dockerfile, as it would only work at runtime (`docker run`).

I got it working like this:

#### Dockerfile

```dockerfile
FROM debian:stretch

WORKDIR /

RUN apt-get update -y
RUN apt-get install -y realpath p7zip-full qemu-user-static git wget kmod sudo python3

RUN git clone https://github.com/guysoft/CustomPiOS.git
RUN git clone https://github.com/guysoft/FullPageOS.git

WORKDIR /FullPageOS/src/image

RUN wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspbian_lite_latest'

WORKDIR /FullPageOS/src

RUN /CustomPiOS/src/update-custompios-paths

WORKDIR /

ADD entrypoint.sh /entrypoint.sh
RUN chmod +x /entrypoint.sh

ENTRYPOINT ["/entrypoint.sh"]
```

#### entrypoint.sh

```bash
#!/bin/bash

modprobe loop

cd /FullPageOS/src

bash -x ./build_dist
```

#### Execution

```nohighlight
docker build -t fullpageos .
docker run -ti --privileged=true fullpageos
```

So I guess all I have to do now is make sure the resulting image file is written to a volume instead of into the container. And I'll probably pass a few `FULLPAGEOS_*` environment variables for customization. Sweet!

closed 2018-02-16 by marians

guysoft, 2018-02-16:
Take a look at the vagrant folder.  If you can PR a docker folder that works for you, I can populate it to all CustomPiOS distros