guysoft/FullPageOS #209: Use of FULLPAGEOS_OVERRIDE_LOCALE should be documented

marians, 2018-02-17:
#### What were you doing?

Setting the locale might have an effect on the behaviour described in #7 .

The script `/src/modules/fullpageos/start_chroot_script` supports the environment variable `FULLPAGEOS_OVERRIDE_LOCALE` to affect this, but it's not explained which value/format is expected to be used.

Attempting to change the locale to German I found that the correct format would probably be

    "de_DE.UTF8 UTF-8"

which is the format contained in `/etc/locale.gen`.

#### Version of FullPageOS?

Current `devel` branch

guysoft, 2018-02-18:
It uses the standard localization of Debian. Which is [documented here](https://wiki.debian.org/Locale#Standard).
It says there:
```
Run locale -a to get a list of the locale names suitable for use in environment variables. Note that the spellings are different from the ones presented in the dpkg-reconfigure list. 
```
We could add that line in the module. Or is there a place you think is better suited?

marians, 2018-02-18:
Adding that in the module would be an improvement.

As I'm not using Debian myself, executing `locale -a` to find the exact string is still a bit of hassle, but doable.

guysoft, 2018-02-18:
@marians You can use a raspberrypi, vagrant. You can even chroot in to FullPageOS in the build process and print it. Next you will say you dont know which package to install ;) cant help with that

closed 2018-02-18 by guysoft

marians, 2018-02-18:
Thanks for the change!

Different issue but related: Is it possible that the package `locales-all` is missing for other locales to be usable?

guysoft, 2018-02-18:
No clue. You are welcome to try and report. If needed we can add it.

guysoft, 2018-02-24:
@marians I can see you are pushing to a new branch there, are you planning a PR?

marians, 2018-02-25:
@guysoft Yes, once I get it working I'll create a PR. Doing this in my spare time though, so time for testing is very limited currently.

guysoft, 2018-02-25:
Yes I know the feeling,  also developing here on spare time only.

commit 6ab0b7fb (2018-02-18): Fixes #209