guysoft/FullPageOS #211: Upgrade from 0.6 to 0.8.1

donisg, 2018-02-21:
In my LAN I run ~150 RPi with FullPageOS 0.6.
I want to upgrade to 0.8.1.
All of them has internet connection.

All RPi has unique config:
1. Different links. RPi-No-1 displays www.sample.com/1 , RPi-No-2 displays www.sample.com/2 and so on.
2. All RPi have different  names. RPi-No-1, RPi-No-2 ...

What is the best scenario to upgrade all 150 RPi and retain configuration?
Connect via SSH to each PRi and run "some upgrade command" would be ok.

guysoft, 2018-02-21:
Wow, that's a lot of Pis!
If I may ask - what are you doing with all of them? Can you send a photo? I was thinking on listing notable uses of FullPageOS in a dedicated issue. It would also help me understand where they all are, since you will have to do some kind of access to the Pis to get them to boot something else.

The short answer is - you can't. FullPageOS 0.6 uses a different Debian release, and there is no way of upgrading between the distros, major things change between each release.

The long answer is - Depends on the level of access. And I will need to understand what exactly is the configuration of this in order to advise.
1. You can make a script that extracts over ssh the configuration of the pi, then get them to boot over the network with a new OS, but only if they are plugged in using Ethernet. Also it depends what version of RaspberryPis you have. [See this information](https://www.raspberrypi.org/blog/pi-3-booting-part-ii-ethernet-all-the-awesome/).
2. You might be able to write a script that takes an SD, extracts the configuration, then flashes the SD card and injects a new configuration, but this assumes you are going to insert each SD card in to a computer and flash it. The hostname and urls used are stored in easy-to-access locations.
Both these methods are ones I haven't tested myself. And again, I need to understand what you are doing there in better detail.

donisg, 2018-02-21:
>If I may ask - what are you doing with all of them?
We are one of the largest bicycles/e-bicycles production in Europe (Lithuania based).
We produce ~500'000 bikes every year. So next time You will see bike on the street, it is possible it was made by our company using FullPageOS :)
Mostly RPi+FPOS are used at assembly lines to display info for workers what they need to do, witch components and tools should be used an so on.

>Can you send a photo?
I can send You photo/video, but only to private (not here). PM me with Your email. I will be able to do this tomorrow.

>I was thinking on listing notable uses of FullPageOS in a dedicated issue.
It is possible, but we will need to discuss.

We use RPi3 only. 
All RPi are connected by Ethernet only and on the same physical network segment (no remote sites, no VPNs or so). I have physical access to all RPi.
All switches are HP 2920 managed, so vlan'ing is possible (if needed).
Since I'm sysadmin, I have root access to any device in the network :)

guysoft, 2018-02-21:
Cool!
My email is guysoft at gmail dot com. Tip - you can get emails from all the git author commits ;)

Ok, so Rpi3 over ethernet should be bootable. So in theory you should be able to ssh to all of them (I am a great fan of fabric, that might do it). Pull all the settings, and then modify them to boot over the network to a custom-made FullPageOS 0.8.1 and you can differentiate between them using mac address or some local setting on the sd. The good thing about it is that once you do this you should be able to upgrade the image any time you want. 
However, I never done this and it will need testing on your side.

And since you have physical access, you could do the upgrade sd script as well. Its also easy to test something like that, esspecially with the [qemu_boot.sh](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/qemu_boot.sh), which with tweaking would let you boot an sd card and see that everything has migrated correctly.

It really depends on you in a way. I never done something like this before and both have pro and cons. I work in a small company I founded and we only have one Pi with a dashboard in the office.

aviyam, 2018-02-21:
Hi,
Having to ssh every one of a 150 RPI seems tedious and boring. You should consider using Ansible (which actuallyuses ssh for its operation). Ansible is great for configuration management.
You should use Ansible to pull all the configurations from your network then you can boot from PXE to install the new OS and then use Ansible to push the configurations back.
Note that this can be done in a single step (Ansible playbook can wait for  the installation to finish).
The problem is that it is a very slow process, as PXE uses tftp which is slow.  You can solve this by  using iPXE over http.
Another approach could be to have a small OS to boot from PXE and from within that OS you use dd to copy an image file onto your RPI. 
To make the image, install a clean install on RPi and copy the SD card to file with dd.

Hope this was clear.

donisg, 2018-02-26:
If I try to boot (same moment) 150 RPi from PXE server, network(1Gbps) will become bottleneck.
Am I right?

guysoft, 2018-02-26:
Are you booting them all at the same time?

donisg, 2018-02-26:
Yes, all same time :(

P.S. Sorry, still not prepared video from production :)

aviyam, 2018-02-26:
I suggest **NOT** to boot all at once. You should try on a small batch (1-5) and see how that goes and how long does it take. Then try a slightly bigger batch and see how long does it take. This could give you an estimation and you can spread your load depending on your bandwidth.

guysoft, 2019-06-24:
Closing if no reply, would value to know what happened here in the end.

closed 2020-10-21 by guysoft