guysoft/FullPageOS #214: MP4 video can not be display

jyy1082, 2018-03-11:
#### What were you doing?
I  installed omxplayer and run:
"omxplayer --win "0 0 799 479" /home/pi/video.mp4" 
the commands shows
[image]
However, there is no video displaying on the screen. 

#### Version of FullPageOS?
0.8.1

guysoft, 2018-03-12:
I am not sure I can support this, this is a raspbian issue.
Its a duplicate of https://github.com/guysoft/FullPageOS/issues/167 . so same answer

jyy1082, 2018-03-12:
We would like to play a video when raspberry pi is booting and then going to chrome. Actually, FullPageOS 0.6 can allow me to implement it, but not 0.8.1. 
Do I need to downgrade to use 0.6 for that?

guysoft, 2018-03-12:
1. Can you provide how you implemented it in 0.6? I can't support code I can't see, and this is an open source project.
2. Who is "we"?

jyy1082, 2018-03-12:
In the 0.6, I added 
omxplayer --win "0 0 799 479" /home/pi/logo.mp4
in /etc/rc.local. 
The video can be played and displayed when booting.
However, I did same in 0.8.1. It does not work. There is no video display
[image]

guysoft, 2018-03-12:
0.6 is based on raspbian jessie. While 0.8.1 is based on raspbian stretch. The boot process from Jessie has changed from sysV to systemd and therefor the console looks diffrent. You will need to make your script support systemd. See internet - for example: https://askubuntu.com/questions/919054/how-do-i-run-a-single-command-at-startup-using-systemd

jyy1082, 2018-03-12:
Thank you! I will try that.

guysoft, 2018-06-11:
Closing because no reply

closed 2018-06-11 by guysoft