guysoft/FullPageOS #216: FullPageOS 0.9.0 RC1 status

guysoft, 2018-03-16:
Hey,
So new 3B+ Pi is out, also several improvements to CustmPiOS (for example removing the network interfaces legacy file. 

So here is a release candidate, please try it out so I know if its good to release:
http://unofficialpi.org/Distros/FullPageOS/nightly/2018-03-15_2018-03-13-fullpageos-stretch-lite-0.9.0.zip

Md5 ``6340cff3925c0ca3894d218bea1df173``

Please test this with new and old Pis and report anything that is a blocker

quist00, 2018-04-11:
Hi, I tried this image on a new 3b+.  

- On the first boot, the help slides were not displayed well. They were compressed veritcally but not horizontally.  After reboot, it seems to display correctly. 

- While the help slides were displayed, any urls I tried to add using the plus button did not show anything. I tried with and without http.   I also tried testing against google and yahoo just to make sure it wasn't my url.   If I replaced FullPageDashboard link with my own link in /boot/fullpageos.txt, then the page would display.

- While the page would now display, every reboot would get stuck on a pop up about chromium not closing correctly and require user intervention.  To get around this, I had to add --incognito to the switches when chromium opens.  However this, switch seems to break the end of that line that reads the url from /boot/fullpageos.txt. Chromium would launch but it would just display a page that said you have entered incognitio mode.  I hard coded my url into the chromium launch string as well and now it displays my URL without user intervention after reboot/power cycling.  

I really appreciate benefiting from all the hard work that has gone into this.  While the current working state is okay for this new books display project we have, the ideal would be to have something more end user configurable like the FullPageDashboard app, so I hope that will continue to function and it is just pilot error on my end.

P.S.  Just in case it matters in replicating any of my experience, I used control-T to pull up a tab to hit a site that would give me my ip address.

guysoft, 2018-04-11:
Ok, 
* so bullet 1 is a known bug in FullPageDashboard.
* bullet 2 is a known bug in FullPageDashboard (https one, and you say it works without it, so its bug there for sure).
* chromium not closing correctly error might be a new chromium behavior, worth investigating.

Yes, I agree FullPageDashboard if bad and should be replaced, but I have litrally ZERO time and the developer who write it is my Co-Founder in our startup and he is even more busy than I am.

@tailorvj said once he might write something instead but it never happened.
This means FullPageDashboard stays until something better is PRed. Its not something I know how to improve.

quist00, 2018-04-12:
Okay, so I dug into the implementation of FullPageDashboard, and see how the SSL pages become an issue within iframes.  For sure that was in play with my tests using google and yahoo, and as more of the web enforces ssl that will only grow.  However, for my own page should not have been an issue unless some library reference was using ssl.  I will have to dig into that Thanks for the info.

guysoft, 2018-04-12:
If you feel like coming up with a solution to it that would be awsome. I have a self-signed ssl proxy written in python if that helps.