guysoft/FullPageOS #217: HyperPixel - 3.5" Hi-Res / High-speed DPI Display not showing anything

MichaelNewham, 2018-03-17:
#### What were you doing?
- Installed FullPage OS on SD Card and booted up on a RPi 3 with a HyperPixel - 3.5" Hi-Res / Hi-speed DPI Display sitting on the 40 pin / GPIO header.
- SSH'ing in works. 
- Seeing 'FullPageDashboard / Chromium' via VNC remote connection works.

#### What did you expect to happen?
After testing display and ensuring it works on another SD Card (with latest Raspbian Stretch), I then installed FullPage OS on another card and after booting up and the usual update /upgrade  - ran the HyperDisplay installer as seen here - https://github.com/pimoroni/hyperpixel

#### What happened instead?
No output to 3,5 inch Display, but can see FullPageDashboard via VNC.
Changes made to config.txt on root of SD card were as follows - 
__________
 - HyperPixel LCD Settings
dtoverlay=hyperpixel
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=800
framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x6f016
display_rotate=2
hdmi_timings=800 0 50 20 50 480 1 3 2 3 0 0 0 60 0 32000000 6

 - Use a basic GPIO backlight driver with on/off support
dtoverlay=hyperpixel-gpio-backlight
__________
#### Was there an error message displayed? What did it say?
No error message, just no display.

#### Version of FullPageOS?
It says 0.9.0, but I'm using the latest FullPageOS 0.9.0 RC1 status.

#### Screenshot(s) showing the problem:
[image]

guysoft, 2018-03-18:
1. Did you use the curl method they give or the setup.sh?
2. Does it work with 0.8.1?
3.
From looking at their source, it looks like they are adding new services:
```
for initfile in ${initlist[@]}; do
    sudo cp ./requirements/usr/lib/systemd/system/$initfile.service /usr/lib/systemd/system/ &> /dev/null
    sudo systemctl enable $initfile
done
```
What is the output of ``systemctl --full | cat``? 
It might indicate if the services failed, or started before/after GUI and that made it fail.

I will note that apart from figuring out what they do, I can't really test this, since I have no HperPixel screen to test with.
I will note that FullPageOS is based on raspbian 2018-03-13, so if it works on Raspbian, It should work on FullPageOS.

MichaelNewham, 2018-03-19:
Tks for quick response!

Okay,
- Tried both curl and setup.sh (from clone of repository on SD card) on 0.9.0 RC1 status - no joy.
see cat of 0.9.0 version attached (9.0 RC1 systemctl --full | cat.txt)
- Tried both curl and setup.sh (from clone of repository on SD card) on 0.8.1 - no joy.
see cat of 0.8.1 version attached (8.1 systemctl --full | cat.txt)

I'll approach the Pimoroni crew  over at https://github.com/pimoroni/hyperpixel - see if they can make any headway..

Again, thanks for your assistance!

Michael

[9.0 RC1 systemctl --full | cat.txt](https://github.com/guysoft/FullPageOS/files/1823292/9.0.RC1.systemctl.--full.cat.txt) 
[8.1 systemctl --full | cat.txt](https://github.com/guysoft/FullPageOS/files/1823348/8.1.systemctl.--full.cat.txt)

MichaelNewham, 2018-03-23:
SOLUTION
Within config.txt on /boot - comment out the original FullPageOS ‘dtoverlay’ (thus making the HyperPixel ‘dtoverlay’ line the default one).

closed 2018-03-23 by MichaelNewham