guysoft/FullPageOS #218: Future Request: auto VPN connection

pompushko, 2018-03-19:
Hello guysoft!

Could you add txt configuration file in root for automatic VPN connection? Like with webpages?

guysoft, 2018-03-19:
You are welcome to implement it yourself, I accept pull requests..
Moreover I have no VPN connection to test against. You could write a guide for a start, then its easy to add a CustomPiOS module.

pompushko, 2018-03-19:
Hello Guy! Could you explain more about CustomPiOS module?

  
  

-- С уважением,Артём Рощеня+375(44)772-00-70

guysoft, 2018-03-19:
See: https://github.com/guysoft/CustomPiOS/wiki/Modules
Again, I suggest you first find a guide that works for you, then you can turn it in to a module.

pide2000, 2018-04-19:
I wrote a description how to switch on vpn and autostart it. Just follow the instructions here https://github.com/guysoft/FullPageOS/issues/187 
It should be easy to put this into a module.

guysoft, 2018-04-21:
You are welcome to implement it yourself, I accept pull requests
I have no time to add these features. Nor time to test them, I manage 5 other distros. You are welcome to send a pull request.

guysoft, 2018-07-22:
Closing if no reply

pide2000, 2018-07-23:
Yeah, close it. At the moment I have no time to implement it. If it is implemented in future we can reference to this issue anyway.

pompushko, 2018-07-23:
bad. very bad :(

closed 2018-07-23 by guysoft

guysoft, 2018-07-23:
@pompushko You are welcome to implement it, I have no time, no money and I am doing this for free. I am not working for you.

pompushko, 2018-07-25:
i could make donation :D