guysoft/FullPageOS #219: Future Request: email notificationb

pompushko, 2018-03-19:
#### What were you doing?

[Please be as specific as possible here]

#### What did you expect to happen?

#### What happened instead?

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2018-03-19:
You are welcome to implement it yourself, I accept pull requests.. I have no time for this.

iGeekuPlay, 2018-06-19:
Can be done with python script and crontab

pompushko, 2018-07-16:
Yeah. i know. But why not add to config file for that ;(

guysoft, 2018-07-25:
@pompushko Because I am not working for you for free. If you want it you can implement this yourself of pay someone. Closing issue if no reply.