guysoft/FullPageOS #222: Setting up Static Wifi address on Scratch

conrad9900, 2018-04-10:
#### What were you doing?

#### What did you expect to happen?

A static address to be assigned to my pi on reboot

#### What happened instead?
No Ip address was assigned

#### Was there an error message displayed? What did it say?
No error message

#### Version of FullPageOS?

2017-11-29-fullpageos-stretch-lite-0.9.0

Firstly In my network, setting the Pi's address on the router side is not possible.
I had wifi working perfectly on Jesse, set it in fullpageos-wpa-supplicant thanks to Guy with no issues.
After reading several of the posted issues I tried the following.
I set the following in dhcpcd.conf:
iface wlan0 inet static
wpa-ssid "Wifi_Net"
wpa-psk "********"
address 192.168.10.13
netmask 255.255.255.0
broadcast 192.168.10.21
I changed nothing in fullpageos-network. The result is no IP address assigned

guysoft, 2018-04-11:
What Is the output of ``ifconfig``?

conrad9900, 2018-04-11:
Will test this later today and post the info

conrad9900, 2018-04-11:
The process i have taken is the following:
1) Fresh install of 
2) Changed dhcpcd.conf to the following:

# Example static IP configuration:
interface wlan0
wpa-ssid="Net"
wpa-psk="1234"
static ip_address=192.168.10.13/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
static routers=192.168.10.21
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

3) reboot, ifconfig shows the following for wlan0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:09:52:08  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
4) I have eth0 active so that i can SSH and apply changes

guysoft, 2018-04-11:
Where do set your wifi username and password?

conrad9900, 2018-04-11:
interface wlan0
wpa-ssid="Net"
wpa-psk="1234"

line 2 & 3

I am just doing this from reading through the issues posted.

guysoft, 2018-04-11:
1. Please reference the issues you are reading.
2. You didnt say where, you said how. What file names?

conrad9900, 2018-04-11:
The only changes i applied were to /etc/dhcpcd.conf and the file looks like the following:

# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Most distributions have NTP support.
option ntp_servers
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate Stable Private IPv6 Addresses instead of hardware based ones
slaac private

# Example static IP configuration:
interface wlan0
wpa-ssid="Peter_Net"
wpa-psk="Profile1"
static ip_address=192.168.10.13/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
static routers=192.168.10.21
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static

# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate Stable Private IPv6 Addresses instead of hardware based ones
slaac private

# Example static IP configuration:
interface wlan0
wpa-ssid="Peter_Net"
wpa-psk="Profile1"
static ip_address=192.168.10.13/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
static routers=192.168.10.21
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static

guysoft, 2018-04-11:
1. Please use ``` before an after when you paste files, I fixes it for you above. You can see how its readable.
2. you wrote inside ``/etc/dhcpcd.conf``:
```
wpa-ssid="Peter_Net"
wpa-psk="Profile1"
``` 
Those lines should be in ``/boot/fullpageos-wpa-supplicant.txt`` and in a different format. Please follow the instrutions there. Also remove those lines, and all other lines you invented, they are breaking the syntax of ``/etc/dhcpcd.conf``.

conrad9900, 2018-04-11:
Ok i have set those in /boot/fullpageos-wpa-supplicant.txt.
Where can i set a static address for wlan0?

guysoft, 2018-04-11:
Please do all I sad above, points 1, 2 and 3. And please paste full files. I am WARNING you one last time.

conrad9900, 2018-04-11:
My apologies, I dont understand what you mean by "Please reference the issues you are reading".

Here is a paste of the FULL file in /boot/fullpageos-wpa-supplicant.txt

```
country=GB

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

# This is only used to configure multiple wifi networks or other advanced wifi features.
# take a look into fullpageos-network.txt instead if you only need basic wifi configuration
# 'man -s 5 wpa_supplicant.conf' for advanced options
#network={
#  ssid="Peter_Net"
#  psk="Profile1"
#}

## You can configure more than 1 wifi networks by adding more 'network' blocks
#network={
#  ssid="Another Wifi"
#  psk="password"
#}
```
Here is also my FULL file for /etc/dhcpcd.conf

```
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Most distributions have NTP support.
option ntp_servers
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate Stable Private IPv6 Addresses instead of hardware based ones
slaac private

# Example static IP configuration:
#interface eth0
#static ip_address=192.168.10.13/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.10.21
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static

# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate Stable Private IPv6 Addresses instead of hardware based ones
slaac private

# Example static IP configuration:
#interface wlan0
#wpa-ssid="Peter_Net"
#wpa-psk="Profile1"
#static ip_address=192.168.10.13/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.10.21
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static

```

guysoft, 2018-04-11:
Replace
```
#network={
#  ssid="Peter_Net"
#  psk="Profile1"
#}
```
with
```
network={
  ssid="Peter_Net"
  psk="Profile1"
}
```
The hashtags symbol (``#``) is called a comment, you need to remove it, this is called to "uncomment". If a line is commented it means the computer ignores what is in it.

Same goes for `` /etc/dhcpcd.conf``, you need to:
1. remove the wpa settings like I said before. I suggest starting with a fresh new file
2. Uncomment

```
# Example static IP configuration:
#interface wlan0
#wpa-ssid="Peter_Net"
#wpa-psk="Profile1"
#static ip_address=192.168.10.13/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.10.21
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
```

What I mean by "Please reference the issues you are reading" is you said in your issue "After reading several of the posted issues I tried the following". That means you are following exiting issues, I want to see which issues you are using because they might be outdated or with information that is wrong.

guysoft, 2018-04-11:
Does the silence means you managed to do something? I would usually give a few days to reply, but since you had two other issues where you just ignored them for over 6 months I'd thought I'd ask now.

conrad9900, 2018-04-12:
Okay i understand what you mean now. I will attempt the changes and go from there.
I do understand the (#) comments, i just forgot to remove. Is there a possibility you will introduce a user friendly interface for network changes? or a single file for those using SSH?

guysoft, 2018-04-12:
1. I have no time to develop a basic interface to replace how Debian fundamentally works. You are welcome to contribute code.
2. How would the user interact with an interface on a system with no keyboard or network connection? The point is you edit the file after you flash the image.
The reason the files are split is that they come from separate software packages. They can also be individually replaced any time by Debian.

conrad9900, 2018-04-12:
Okay, i understand, it is just difficult to keep up with the changes and the syntax with each debian release.
Thanks for the help. I can consider this issue resolved.

conrad9900, 2018-04-12:
```

closed 2018-04-12 by conrad9900