guysoft/FullPageOS #223: Wrong Display Resolution

metalRudge, 2018-04-11:
#### What were you doing?

I just installed fullPageOS version 0.81 on my rPi model 3 B. In an earlier version this issue was not present. After installing and setting the fullpageos.txt to the website I wanted to display I plugged the SD card back into the pi and booted but the issue was present even after a new installation. 
I tried to modify the overscan and force hdmi in different modes( namely 1 & 2 ) but to no avail. 

#### What did you expect to happen?
I expected it to display the correct resolution, the one rPi gets from the HDMI output. This issue was not present in a earlier version of this software.

#### What happened instead?

The resolution Fullpageos displays at is wrong. The borders arent alligned with the display borders. This was tested with 2 different monitors aswell as a Samsung TV.

#### Was there an error message displayed? What did it say?

There was no message.

#### Version of FullPageOS?
0.81

#### Screenshot(s) showing the problem:

I will attach if its necessary.

guysoft, 2018-04-11:
All the hdmi stuff os maintained by raspbian. Did you try 0.9.0 which ships the latest raspbian?

metalRudge, 2018-04-12:
I will try today, thanks

metalRudge, 2018-04-13:
I tried the nightly build and changed some hdmi settings in the config file but it didn't help. I am now running version 0.9. Is there anything else I can try ? Because earlier, in version 0.6 it worked without adjusting the settings. Now I have an overlap on all different sides.

guysoft, 2018-04-13:
Its likely something changed in Raspbian, you can try see if Raspbian helps.
You can also try disabling graphical acceleration which was enabled by default on 0.8.

marians, 2018-05-11:
I think I am running into this problem as well. I was working with an image I flashed a couple of months ago, not sure what version. Then I did an `apt-get update` and `apt-get upgrade`. After a reboot, the browser content appeared scaled, and the right and lower boarder were not visible. Via VNC however I still see the unclipped view.

Are there any news on this? Can I deliver any diagnostic information that could help solve the problem?

marians, 2018-05-21:
Little update on my previous comment: I changed the GL driver settings via raspi-config and the picture now shows up in the right scale, correctly fitting the monitor. Only problem is that I don't know what setting I changed it to. However, if somebody else runs into this problem and has the nerve to experiment interactively, give it a try. Thanks @guysoft for the hint!

guysoft, 2018-05-21:
Cool, shall I close the issue?

marians, 2018-05-21:
For my part you can close the issue, especially as it's a bit unclear when this happens. It seems to depend on the monitor. If you agree that the aim should be to have this work reliably and "hands-free" (without administrator interaction), I would re-open this when I have more information.

closed 2018-05-21 by guysoft