guysoft/FullPageOS #227: Blank Pages

gleneatkinson, 2018-04-30:
#### What were you doing?

trying to load any external address either with url or ip address
#### What did you expect to happen?
Page would open and display

#### What happened instead?
a blank white page would display

#### Was there an error message displayed? What did it say?
no errors

#### Version of FullPageOS?
0.8.1 and 0.9.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:
[image]

[If applicable. Always include if unsure or reporting UI issues.]

gleneatkinson, 2018-04-30:
Please note this only occurs in full page dash board if I don't use the dashboard a page will load fine

guysoft, 2018-04-30:
FullPageDashboard board is dead an unsupported, its only there because no one would support it or write anything better (which should be really easy considering that https is not working).

If it does not do what you want don't use it.

closed 2018-04-30 by guysoft