guysoft/FullPageOS #229: Feature Request

asdf1nit, 2018-05-08:
Looking into piServer for deployment and management of your OS and thought maybe you would consider either

1. Creating a tar.xz of the os for easy importing into piServer either by downloading or the URL option
or
2. Some documentation on creating this

guysoft, 2018-05-12:
Interesting. Would help #211 .
I have no real way to test this though.

What I can suggest is a script that exports the OS in to a ``tar.xz`` file.
You could run on an image something along the line of (you will need [CustomPiOS](https://github.com/guysoft/CustomPiOS) installed and ``common.sh`` from it in path.

```
mkdir /tmp/mount
source /path/to/CustomPiOS/src/common.sh
mount_image /path/to/extracted/fullpageos.img 2 /tmp/mount
cd /tmp/mount
tar cfJ /tmp/fullpageos_piserver.tar.xzt
```
You might need to unmount ``/dev`` and ``/proc`` first I just wrote this and never ran it.

Alternatively I would gladly accept a CustomPiOS module that would do this. Would also be useful for RealtimePi.

asdf1nit, 2018-05-18:
I haven't looked into CustomPiOS as I've just run across your repo but, I think that I'm going to give this a shot. I have no where close to the 150 pi that issue #211 has but deploying 20 is enough for me to not want to mess with frequent, even twice yearly updates to my pi's.

If you want to close this issue go ahead, and thanks for the starting point. Once I've got something worked out if it benefits your project I'll submit a pull.

guysoft, 2018-06-09:
See https://github.com/guysoft/CustomPiOS/commit/4537a1eb332d8a562e9e5174a340ee172582b94a

If you export the filesystem to a tar.gz it should work.

asdf1nit, 2018-06-26:
Not quite understanding the use of "export fullpageos /". I'm fairly new to managing linux systems. I'm a sysadmin mostly windows and mac but my google fu is failing me at the moment for the process to accomplish this. 

I've added common.sh to my path but it's not acting correctly, to my knowledge. If I call it with parameters it just returns to the cmdline with no output.

As for the script you mention and posted earlier  having an image that's bootable with my tweaks is the problem I'm having. I have the current running build that I'd like to use but how would I image that while it's running?

Maybe I'm going about this the wrong way. Could you point me in the direction of some documentation so I can learn this?

guysoft, 2018-06-26:
You need to source common.sh. Not add it to your path.

TakuyaSan420, 2019-05-22:
Anything new on this?
Trying for days to make it PiServer "compatible"
Thanks

guysoft, 2019-05-22:
@TakuyaSan420 OP said its beyond their scope, I have no time for this. I gave an explanation how this might be done, if you feel like trying.

asdf1nit, 2019-07-04:
I never went down this path but I did find a solution. I started using cockpit https://cockpit-project.org/ which brings all of my pi's into one management interface, with a little setup overhead.

It can sync your users and login info as well as run updates relatively easy.

In stretch you have to add the debian backports repository but in buster it is now in the testing repo which doesn't need any additional configuration to work. I haven't been able to add the program to a module successfully as it has some dependencies that don't work in a chroot.

closed 2019-07-04 by asdf1nit