guysoft/FullPageOS #234: Timezone should be configurable with config files

mikkotikkanen, 2018-06-11:
Timezone should be configurable with the config files, similar to network settings.

Currently, using FullPageOS for graph dashboards (in order to set timezone correctly) is bit tedious with multiple screens as for every installation/update you would need to separately SSH to each box or dig out keyboards to manually go through every box. This would be far superior with just copying the same config files on every install/update, much like with network settings.

guysoft, 2018-06-11:
What config file are you copying?

#63 lets you control the timezone for your own builds

You can ssh once to each machine and run:
```
echo "$FULLPAGEOS_OVERRIDE_TIMEZONE" > /etc/timezone
dpkg-reconfigure -f noninteractive tzdata
```

This should be permanent.

mikkotikkanen, 2018-06-11:
I'm (or anyone else) installing the ready-built images, or if update is needed just overwriting the existing install with new write to the SD card. The boxes are/would be used in multiple locations in multiple countries so SSHing to every new one isn't really feasible. Having the config file like `fullpageos-wpa-supplicant.txt` ready-made which could be copy-pasted on any computer as the SD card has been written would be the ideal solution for anyone doing similar things.

guysoft, 2018-06-11:
Just to understand your usecase - you personally have multiple devices around the world? Or are you saying hypothetically?
What exactly are you doing?

mikkotikkanen, 2018-06-11:
Not me personally but the company that I work for, offices in 4 different countries, thus the "distributed" nature of problem. Currently we are using FullPageOS for devops screens that show graphs & statistics about the production stacks for each location/team specifically and for the shared systems, however, one of the issues is that the timestamps in the screens don't match the locations as timezones default to UTC.  Currently only thing that they need to do for new screen is to just install FullPageOS to new SD-card, copy the network settings file and power the thing up. If there would be similar file for other settings as well, they could also add that, specify their timezone and get correct graphs.

guysoft, 2018-06-11:
ok, Cool.
I have zero time to implement this, so a pull request would be appreciated. You can probably also pay a developer outside the company if you don't have the knowledge in-house, and I can help you understand where to put this.
I am building this in my own free time, I am not paid like you, and I barely have time to maintain this distro along with my other 7, which are all FOSS I do in my spare time.

A timezome file module would make sense in CustomPiOS.

asdf1nit, 2019-07-04:
This doesn't work in debian stretch and later. They have reconfigured how tzdata is read.

Found a post describing it here 
[https://www.illucit.com/en/linux/timezone-in-debian-9-stretch/](url)

Modified the following in FullPageOS/src/modules/fullpageos/start_chroot_script:

`#override timezone
if [ "$FULLPAGEOS_OVERRIDE_TIMEZONE" != "default" ]
then
    echo "$FULLPAGEOS_OVERRIDE_TIMEZONE" > /etc/timezone
    dpkg-reconfigure -f noninteractive tzdata
fi
`
to

`#override timezone
if [ "$FULLPAGEOS_OVERRIDE_TIMEZONE" != "default" ]
then
    ln -fs /usr/share/zoneinfo/"$FULLPAGEOS_OVERRIDE_TIMEZONE" /etc/localtime
    dpkg-reconfigure -f noninteractive tzdata
fi`

This works for me but I haven't tested it on any other time zones as my builds take about 40 min.

guysoft, 2019-07-07:
@asd1int A pull request would be welcomed

asdf1nit, 2019-07-07:
I'll do a bit of testing this week and if there's no issues with setting it this way I'll submit it.