guysoft/FullPageOS #235: New Build Not Booting Pi 3 B+

conrad9900, 2018-07-05:
#### What were you doing?

Writing Image to new 64GB SanDisk SD Card

#### What did you expect to happen?

To boot to FullPageOS

#### What happened instead?

Boots past Rainbow landing page, Black Screen after

#### Was there an error message displayed? What did it say?
No Message, only black screen,
Tried a different SD card, still no boot. 
Formatted the SD card first also

#### Version of FullPageOS?

2018-06-27-fullpageos-stretch-lite-0.10.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

closed 2018-07-05 by guysoft

jyy1082, 2018-08-30:
I just tried "2018-04-22_2018-04-18-fullpageos-stretch-lite-0.9.0.zip" on 3 B+, but it does not work. It can not boot, just black screen.

guysoft, 2018-08-30:
@jyy1082 try the non-acceleration version at:
http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip

There are reports with 3B+ acceleration not working on some models, the fact you are based in China might give a good hit this is the issue, because you likely have access to much more diverse hardware there, which I can't test.

jyy1082, 2018-08-30:
The non-acceleration version works. 
Could you tell me how should I disable GPU when use CustomPiOS to build a non-acceleration image?
Thank you!

guysoft, 2018-08-30:
The non-acceleration version just disabled the acceleration in the gui module.
Its variant is here: 
https://github.com/guysoft/FullPageOS/tree/devel/src/variants/no-acceleration

You can read about it here: https://github.com/guysoft/FullPageOS#building-using-vagrant

jyy1082, 2018-08-30:
How about I build from Debian and just set "[ -n "$FULLPAGEOS_INCLUDE_ACCELERATION" ] || FULLPAGEOS_INCLUDE_ACCELERATION=no" in this file:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/config
?

guysoft, 2018-08-30:
You can also write it from scratch. The varant and module system are there so you can easily keep your own code separate. You could also add a ``jjos``folder and append it to the main ``config`` file, if you are planning to maintain code against FullPageOS, which I am guessing you are. Read about writing CustomPiOS modules.

jyy1082, 2018-08-30:
Got it. Thanks so much for your support!