guysoft/FullPageOS #237: Trying to run an html site but it is being choppy on raspberry pi

samijandali, 2018-07-21:
#### What were you doing?

I have a Raspberry Pi 3 B, I am running this site pitastreetfood.com/tv-slideshow-page on fullpageos

#### What did you expect to happen?

I expect to have a slideshow of images and videos and a marquee on the bottom to be smooth

#### What happened instead?

Everything is laggy and choppy, it works fine on my PC

#### Was there an error message displayed? What did it say?

No

#### Version of FullPageOS?

0.9.0, latest version

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2018-07-21:
Use ``fullpageos.txt`` in the /boot folder and not FullPageOSDashbaord.

samijandali, 2018-07-21:
I used fullpageos.txt and i changed the link there to my link. I followed the instructions on the ReadMe.

guysoft, 2018-07-21:
And?

samijandali, 2018-07-21:
Thats what I have done originally in my first post. Replaced my link with the one in fullpageos.txt. can you test it please?

guysoft, 2018-07-22:
Hey, I created a variant of FullPageOS that has graphical acceleration disabled, so you can test that:

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip

md5: [2197ebd78f65c17e4155e5bf93e7af55](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip.md5)

guysoft, 2018-07-22:
No free Pi at the moment. If I boot it up and its choppy there is nothing I can really do. You might need to optimise the page or try a Pi 3B+