guysoft/FullPageOS #241: 64GB SD card, Etcher, no IP after boot

attila-kocsis, 2018-08-06:
I have a 64GB Sony SD card, on the latest nightly build, flashed with Etcher. After the boot process the RPI does not get an IP address from DHCP. Tested on 2 different RPIs.

However if I do the same on the 16GB SD card (all the configuration is identical) it works as intended. Any tips why this is happening?

guysoft, 2018-08-06:
Does your card model appear in the

attila-kocsis, 2018-08-06:
Not exactly, it is a SR-64UY2. There are SR-64UY and SR-64UY2A/TQ in the list.

guysoft, 2018-08-06:
64UY2A/TQ should be close enough.
Does it work with normal latest Raspbian?

attila-kocsis, 2018-08-06:
Will check and get back with the results.

attila-kocsis, 2018-08-06:
I flashed the latest Raspbian, it boots and after I configure the same network it gets an IP address.

guysoft, 2018-08-06:
Ok, what version of FullPageOS are you using?

attila-kocsis, 2018-08-06:
The latest nightly (0.10.0)

attila-kocsis, 2018-08-07:
OK, will test it and get back with the results.

attila-kocsis, 2018-08-08:
Tested it with the latest stable build (0.9.0) and the results are the same. RPI boots, but does not get IP address and shows ERR_INTERNET_DISCONNECTED message in Chromium.

guysoft, 2018-08-08:
FullpageOS is nearly the same as raspbian, there should not be any change n how internet connection is set compared to raspbian. The only major difference is that you can edit wpa-supplicant (aka ``fullpageos-wpa-supplicant.txt``) directly and not set it via the GUI. 

You did not explain how you set your wifi using raspbian - how are you doing that?

guysoft, 2018-08-08:
Ok, so it seems you are setting the internet differently in raspbian than in FullPageOS. So I can't really compare to that. Just confirm is not a hardware issue.
You are saying you set ``fullpageos-wpa-supplicant.txt`` using a different SD card, and that it failed to connect.
Did you try the non-accelerated version?
Will note at this point I am making pure guesses, I have no idea what is going on.

attila-kocsis, 2018-08-09:
Ok, thanks for the help so far. I won't invest time on this anymore as it is not a blocker for me now, will use 16GB SD cards.

closed 2018-08-09 by attila-kocsis