guysoft/FullPageOS #243: Bricks SD card fairly frequently

rsouthgate, 2018-08-20:
#### What were you doing?

5 Pi 3s running 0.9. Have been using the FullPageOS for about 2 years. The Pis are dashboards in kitchens and quite probably are getting abruptly power cycled by cleaners, kitchen hands basically anyone who doesn't know any better. They are failing to boot every month approx.

#### What did you expect to happen?

I guess I'd expect that to occasionally corrupt the SD card. It just seems to happen more frequently than I would have thought!

#### Version of FullPageOS?

0.9.0

I would like to be able to configure the FullpageOS to be a Read-only file system. I got all excited that I could just run this script https://www.raspberrypi.org/blog/adafruits-read-only/ but the Pis boot to a standard desktop instead of chrome.

I then tried following this: https://www.raspberrypi.org/forums/viewtopic.php?p=1044815 and while it all booted back up the filesystem was still writeable (something to do with the bootloader not loading the initramfs...) So I'm desperately reaching out to see if you might have a recommended approach to protecting the SD card from power cycling?

guysoft, 2018-08-20:
Strange, I have 0.9.0 running for a few years and had no corruption. Also are quite a few reports of people running it in places where this did not happen.
What power supply are you using? And what SD card?

Also, I can see the script (assuming you use [this script](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/read-only-fs.sh#L269) ) plays with lightdm and might be breaking the session setup. Also there are a bunch of switches so I don't know what you selected.
It might be a good idea to add a read-only module to [CustomPiOS](https://github.com/guysoft/CustomPiOS), PRs are welcome..

xvilo, 2018-09-12:
SD Cards aren't made for continuous use by for example an OS. That is why they tend to break fast (cheaper ones more frequently). Try buying a more expensive one from a known vendor like SanDisk (Extreme)

rsouthgate, 2018-09-12:
It's not that the SD card gets permanently ruined though... just corrupted and requiring re-flashing.

I've managed to implement a temp overlay file system using this setup: https://gist.github.com/dzindra/a8e2083a7f037ca244cf70d100c96656

Also added ro to the boot partition in /etc/fstab.

On two of the Pis I setup this way I also had to add this to the cmdline.txt to avoid a kernel panic:
dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 (Not an actual fix, just disabling fast interrupts: https://www.osadl.org/Single-View.111+M5c03315dc57.0.html). 

And on some I had to uncomment the line in config.txt that boosts the hdmi output:
config_hdmi_boost=4

Note: Test rig (HDMI cable, power adapter etc) was same for all, so what this highlighted to me was that the actual internal performance of supposedly identical Pis can vary significantly.

guysoft, 2018-09-13:
The script @rsouthgate [mentions](https://gist.github.com/dzindra/a8e2083a7f037ca244cf70d100c96656) is cool, and would fit nicely as a [CustomPiOS module](https://github.com/guysoft/CustomPiOS/blob/314f9c42dd0fee742d10c53228f5adb97b7a171c/src/modules/read-only-overlay/start_chroot_script). 
But it rebuilds initramfs, for only one kernel, and uses that initramfs only. It needs to work on both kernels to be shippable

mikopia, 2021-02-12:
Was there a solution to this? I had another one crash last night. I think it said superblock error. And it's a Sandisk card too.

guysoft, 2021-02-14:
It seems that in more recent builds over the past last 3 years this has become much more stable. Either SD card quality got better or the Rpi image.

akfsp, 2022-04-21:
As we are concerned by this I am looking into enabling read-only filesystem.
Apparently all it needs is to enable it through `raspi-config` - as described in this guide: https://learn.adafruit.com/read-only-raspberry-pi/ 
-> Has anyone any long-term experiences with that?

MKaidan, 2022-04-26:
> As we are concerned by this I am looking into enabling read-only filesystem. Apparently all it needs is to enable it through `raspi-config` - as described in this guide: https://learn.adafruit.com/read-only-raspberry-pi/ -> Has anyone any long-term experiences with that?

It seems to be hit or miss for me. We have 13 RPis setup with FullPageOS and around 5 or 6 have been solid since setting up read-only filesystem. I can't seem to find consistency even with read-only.