guysoft/FullPageOS #244: RAM Full?

a2th3g, 2018-08-21:
#### What were you doing?
Running multiple setups with a touchscreen, and all start behaving  **extremely** slow after about 15 hrs of use.

#### What did you expect to happen?
Need stable runtime, or auto restart at pre determined times
#### What happened instead?

#### Was there an error message displayed? What did it say?
no errors just slow performance
#### Version of FullPageOS?
Running for about a month....not sure version.
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2018-08-21:
1. Please provide the version, you message include the instructions how.
2. what page are you running?

a2th3g, 2018-08-29:
The site is “Monday.com”

Linux fullpageos 4.14.34-v7+ #1110 SMP

My goal at the moment is force a reboot based on time. And update the load/splash screen.

guysoft, 2018-08-29:
Can you please answer my questions, I can't help you otherwise.
You did not answer:
1. Please provide the version (of FullPageOS), you message include the instructions how.
3. Additionally want version of RaspberryPi are you using?

guysoft, 2018-11-20:
Closing if no reply

closed 2019-03-10 by guysoft