guysoft/FullPageOS #245: Default background color?

Jaxel, 2018-08-23:
Is there a way to set the browser to have a black background by default? Instead of white?

guysoft, 2018-08-23:
I don't understand what you mean by 'browser background color'. The browser is by default in kiosk mode and none of its interface is visible. If you are referring to the web page, you will need to select a black page and NOT use FullPageDashboard.

Jaxel, 2018-08-24:
All browsers have default CSS rules. Default backgrounds, default fonts, default font colors. Many browsers even let you edit these default CSS rules.

guysoft, 2018-08-24:
The running browser is chromium. So I guess you just need to look how it is done in chrome.

guysoft, 2018-08-26:
``/home/pi/.config/chromium`` afaik

Jaxel, 2018-08-26:
Apparently, Chrome removed the ability to edit the default CSS colors back in 2015...

In IE/Edge, you can still do it by typing "internet properties" into search and heading to the appearance section.

guysoft, 2018-08-26:
You could probably mimic this with a greesemonkey userscript