guysoft/FullPageOS #247: Not working with Official 7" Touchscreen...

Jaxel, 2018-08-25:
I used Etcher to install "2018-04-22_2018-04-18-fullpageos-stretch-lite-0.9.0" to my SD card. Then edited the "fullpageos-wpa-supplicant.txt" file to connect to wifi. Plugged everything in and turned it on. On first boot, it ran the bootup process on the 7" touchscreen; then went to the "resizing storage" screen and rebooted. After that, my screen is completely blank; it's just a lit up black screen.

If I plug a monitor in using the HDMI cord and boot the system, it all works fine. However, unplugging the HDMI cord and booting with the official 7" touchscreen plugged in gives me nothing.

I was able to get it working using "2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0"; but I don't want to use that version because jessie has issues with LCD rotation and touchscreens. I have been unable to get any stretch version working with the touchscreen.

guysoft, 2018-08-26:
Can you try rhe no acceleration test image in nightly? 
http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip
md5:
http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip.md5

Jaxel, 2018-08-26:
I will be streaming later today, so I won't have time to test today. Maybe
tomorrow.

guysoft, 2018-08-26:
Are you streaming using FullPageOS?

Jaxel, 2018-08-26:
I use it as a kiosk for my chatroom. https://imgur.com/gallery/tAuYC

I'll be streaming in about an hour or two if you want to check it out...
https://twitch.tv/8wayrun

guysoft, 2018-08-28:
I guess this means gpu needs to be disabled in the touchscreen display. I will
release two images next release,  and we can document when to use each.

pslogic, 2018-08-30:
Just an information note... with the built-in stand orientation of the official rpi touchscreen case, the screen output ends up being upside down. it may be worthwhile add some "heads-up" guidance to the touchscreen display distro release notes.

guysoft, 2018-08-30:
You can write here what you think should go. And I will paste it in, or you are welcome to send a PR to the readme page.

GekoCH, 2018-11-28:
the new "2018-11-20_2018-11-13-fullpageos-stretch-lite-0.10.0" nightly version does not work.
During the first boot after flashing the SD-Card the boot sequence gets displayed on the 7" RPI screen. But after the boot sequence finishend I get a black screen.
After rebooting the screen stays black even during the boot sequence...

Any Ideas?

guysoft, 2018-11-28:
@GekoCH Try the non-acceleration version and report:
http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/

steampixel, 2018-11-30:
Same problem here :-(

steampixel, 2018-11-30:
I did a `sudo raspi-config`
Than I first switched to Advanced Options -> Memory Split and increased the GPU memory to 256
Than I launched the Advanced Options -> GL Driver panel and switched to Legacy mode
Now it works
A few infos about my system:
`
pi@fullpageos:~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Rev 1.2

pi@fullpageos:~ $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
# enable raspicam
start_x=1
gpu_mem=256
#dtoverlay=vc4-kms-v3d

lcd_rotate=2

`

peebles, 2023-10-15:
> The workaround is to use the non-acceleration image.

What exactly is the non-acceleration image?  nightly 2023-05-12 or nightly 2023-05-09 (from rpi Imager)?

Or do I need to build an image?

In Stable, in respi-config, if I enable Legacy mode, the last thing I see on the screen is the Network Time Sync message.  It stays there, without blanking the screen, seemingly frozen.

pauldogg, 2024-02-11:
Any chance we can get the 7" touch screen to be supported?
This would make for a really nice display.

Tested on all versions available on pi imager 1.8.5
Tested with several different pi's

Thanks

guysoft, 2024-02-11:
The offical pi display worked for me a few months ago - what is not working for you?

ItsMeNJC, 2025-07-13:
> Not sure if this is still a problem for people... But if anyone finds this because they are having an issue! What solved it for me was to uncomment in `config.txt`:
> 
> ```
> dtparam=i2c_arm=on
> dtparam=i2s=on
> ```
> 
> Display + Touch all working.
> 
> > fullpageOS bullseye armhf lite 13.0

Thanks for taking the time to add this comment. It worked for me using a generic 1280x400P display on Pi Zero 2W.