guysoft/FullPageOS #250: Can't connect to WiFi

crood58, 2018-09-11:
I have been trying to get FullPageOS to run on a Raspberry Pi 3 and have had no success. I tried at the location were it is being installed and had no luck. I brought it home and still no luck. Does anyone happen to know what could be going wrong? I have provided the contents of the WPA Supplicant below. Please note that I changed the password to a fake one for security reasons. 

# Use this file to configure your wifi connection(s).
#
# Just uncomment the lines prefixed with a single # of the configuration
# that matches your wifi setup and fill in SSID and passphrase.
#
# You can configure multiple wifi connections by adding more 'network'
# blocks.
#
# See https://linux.die.net/man/5/wpa_supplicant.conf
# (or 'man -s 5 wpa_supplicant.conf') for advanced options going beyond
# the examples provided below (e.g. various WPA Enterprise setups).
#
# !!!!! HEADS-UP WINDOWS USERS !!!!!
#
# Do not use Wordpad for editing this file, it will mangle it and your
# configuration won't work. Use a proper text editor instead.
# Recommended: Notepad++, VSCode, Atom, SublimeText.
#
# !!!!! HEADS-UP MACOSX USERS !!!!!
#
# If you use Textedit to edit this file make sure to use "plain text format"
# and "disable smart quotes" in "Textedit > Preferences", otherwise Textedit
# will use none-compatible characters and your network configuration won't
# work!

## WPA/WPA2 secured
network={
  ssid="RDS-Basement"
  psk="fakepassword"
}

## Open/unsecured
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#}

## WEP "secured"
##
## WEP can be cracked within minutes. If your network is still relying on this
## encryption scheme you should seriously consider to update your network ASAP.
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#  wep_key0="put password here"
#  wep_tx_keyidx=0
#}

# Uncomment the country your Pi is in to activate Wifi in RaspberryPi 3 B+ and above
# For full list see: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
#country=GB # United Kingdom
#country=CA # Canada
#country=DE # Germany
#country=FR # France
country=US # United States

### You should not have to change the lines below #####################

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

Thanks in advance for the help. 

Chris

guysoft, 2018-09-12:
1. What version of FullPageOS are you using?
2. Are you using 2A or more power supply?

crood58, 2018-09-12:
@guysoft,

I am using the 4-8-18 stable version, which is the latest I can see. Also, using a 2.5A power supply.

guysoft, 2018-09-12:
14-8-18 is the baseimage version, not the version of FullPageOS.
When you created this issue you ignored the instruction:
```
#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]
```

I have no way to find out why your instance does not connect. You can try normal raspbian lite (not the desktop version) and see if that works, it might not be a FullPagOS issue.

crood58, 2018-09-12:
So where do I get the full page os image then?

guysoft, 2018-09-12:
https://github.com/guysoft/FullPageOS#where-to-get-it

crood58, 2018-09-12:
This is the one I downloaded. Should I of download a nightly build?

http://unofficialpi.org/Distros/FullPageOS/2018-04-22_2018-04-18-fullpageos-stretch-lite-0.9.0.zip

guysoft, 2018-09-13:
That is version ``0.9.0``.
I am saying you should try raspbian lite and see of it has the same problems here:
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-06-29/2018-06-27-raspbian-stretch-lite.zip

guysoft, 2019-06-23:
Closing, no reply. Version 0.10.0 is out too.

closed 2019-06-23 by guysoft