guysoft/FullPageOS #259: SSL fail : enable ntp-wait so we can browse HTTPS website

olaulau, 2018-10-18:
#### What were you doing?
opening FullPageOS directly on a valid HTTPS website

#### What did you expect to happen?
load and display the website

#### What happened instead?
chromium refuses to load the website, arguing that the certificate is invalid. surely due to the fact that raspberry doesn't have a RTC.

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?
0.9.0

#### Screenshot(s) showing the problem:

guysoft, 2018-10-18:
1. Can you supply a current workaround or something so I can understand what you want change?
2. Does this work on a normal Rasbian Lite installation?

olaulau, 2018-10-18:
1 : workaround : I had to disable http to https redirection so that the pi can load the website the insecure way. but you have to own the website to do that.

2 : raspbian lite : it doesn't have any GUI, but i never had the problem on normal raspbian.

http://manpages.ubuntu.com/manpages/cosmic/man8/ntp-wait.8.html

brainstomp, 2018-10-23:
Having the same problem with the build dated 2018-09-13 named 2018-06-27-fullpageos-stretch-lite-0.10.0

I can display HTTP sites but not HTTPS. I haven't found a workaround.

Ran 
sudo apt-get update 
sudo apt-get upgrade

rebooted still no luck.

Raspberry Pi 3 B+

guysoft, 2018-10-23:
@brainstomp 0.10.0 is a nightly build, have you tried 0.9.0?

guysoft, 2018-10-23:
@brainsstomp, wait, are you using FullPageDashboard? Thay does not support https.

guysoft, 2018-10-23:
@brainsstomp, wait, are you using FullPageDashboard? That does not support https.

brainstomp, 2018-10-23:
Yea I was using the dashboard. I will try without it and see what shakes.

Thanks!

MatrixStefan, 2019-02-21:
Hi - I've managed to connect to https URLs outside FullPageDashboard, but obviously this eliminates the ability to cycle tabs.

tailorvj, 2019-02-21:
@MatrixStefan We are working on an experimental rotation system right now. An initial version should be released soon. It will have built in playlist support. Please [subscribe to the experimental features group notification list](http://eepurl.com/ghHv4n). 

Cheers,
Tailor