guysoft/FullPageOS #262: Unable to set video resolution to 60Hz / 1080p

Prinzessor, 2018-11-13:
#### What were you doing?

I have four fullpageOS raspis connected to TV.

#### What did you expect to happen?

I set hdmi_mode=82 and hdmi_group=2 in config.txt
I expect to have the pis running in 1080p/60Hz.
This is working for the default raspian distribution.

#### What happened instead?

When the pis are booting, the TVs are showing 1080p/60Hz. A soon as the graphical interface is running, the TV switches to 1080p/50Hz. With 50Hz they are flickering ...

#### Was there an error message displayed? What did it say?
No

#### Version of FullPageOS?
Version 0.9.0

guysoft, 2018-11-13:
Did you try using the no-acceleration variant?
http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip

md5: [`2197ebd78f65c17e4155e5bf93e7af55`](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip.md5)

guysoft, 2018-11-13:
Its a release candidate, it will be a release if no one tests and reports.
It won't if you test and find an error.

There are currently not enough testeds.

Prinzessor, 2018-11-13:
I will give it a try the next few days!

Prinzessor, 2018-11-21:
Yesterday I just got to test it, and it works perfectly with the no-acceleration image! Thank you!

closed 2018-11-21 by guysoft