guysoft/FullPageOS #263: chromium no longer restarts automatically after being closed

gbrakcounty, 2018-11-19:
#### What were you doing?
After the fullscreen chromium loads, pressed CTRL+W to close it.  Back in version 0.8.1, chromium would automatically reload if i closed it using CTRL+W.  Now in version 0.9.0, it does not.  This is problem since i will be using this for public kiosks.  If someone happens to close the window, intentionally or accidentally, i will have to remotely reboot the device.  Not efficient way of doing things.  Now i am not sure if that feature was intentional in the previous version or not, but an excellent feature nonetheless.
[Please be as specific as possible here]

#### What did you expect to happen?
chromium should have re-opened after being closed.  worked fine in version 0.8.1
#### What happened instead?
nothing, black screen, mouse cursor.
#### Was there an error message displayed? What did it say?
none
#### Version of FullPageOS?
issue occurs with 0.9.0, was not present in 0.8.1
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2018-11-20:
Ok, looking at the code it should have failed in 0.8.0 too. However this is a valid regression for older versions.
I know what to fix - add a loop on ``start_chromium_browser``. Will be able to test tomorrow.

gbrakcounty, 2018-11-20:
Good thing I made an img file of a working kiosk set up. Thanks for the prompt reply. I'll take look at making a loop in that file.

From: Guy Sheffer
Sent: Tuesday, November 20, 9:09 AM
Subject: Re: [guysoft/FullPageOS] chromium no longer restarts automatically after being closed (#263)
To: guysoft/FullPageOS
Cc: Brakatselos, George, Author

Attention: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.
Ok, looking at the code it should have failed in 0.8.0 too. However this is a valid regression for older versions.
I know what to fix - add a loop on start_chromium_browser. Will be able to test tomorrow.