guysoft/FullPageOS #264: chromium keyboard shortcuts pose security risk to kiosk while in public areas

gbrakcounty, 2018-11-21:
#### What were you doing?
Chromium Keyboard shortcuts are still active while in kiosk mode.  New windows and tabs can be opened which poses a security risk, allowing a public user to tamper with Chromium.  Tried using xmodmap to disable CTRL ALT keys but the setting does not apply to Chromium.  Even tried using an extension to modify keyboard shortcuts, was able to disable or reroute most of them but there was still a way to open a new window CTRL+SHIFT+N.  Not sure if this is an issue with FullPageOS, seems more a chromium issue, but since this is a kiosk project, the availability of keyboard shortcuts and inability to manage them properly to secure the kiosk kind of defeats the purpose.

#### What did you expect to happen?

#### What happened instead?

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?
0.8.1
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2018-11-21:
I can add this as a feature of ita avilable in chromium, preferably as a commandline flag

gbrakcounty, 2018-11-21:
> Attention: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.

I managed to reroute the shortcuts that could open new tabs, including history, downloads, print window, bookmarks, bookmark bar.  The only shortcut I left available was ctrl+w and ctrl+shift+q because that can be useful for troubleshooting.  The extension I used is shortkey, although the extension itself proved useless for disabling shortcuts, it did enable chromiums built in keyboard shortcut manager, which I used to reroute the commands.  So far from what ive seen chromium does not support disabling of shortcuts.

bcherup, 2018-12-15:
@gbrakcounty 

Do not use standard keyboards in public areas -  Modify the existing keyboard or buy one made for public usage (kiosk type).

https://www.aliexpress.com/store/product/Custom-40-keys-IP65-fully-sealed-USB-PS2-RS232-interface-metal-keypad-keyboard-with-numbers-letters/1360552_32319513912.html?spm=2114.12010612.8148356.21.79aa519epmkg9b

gbrakcounty, 2018-12-17:
Oh ok good to know thanks. I disabled pretty much all shortcuts that can access the file explorer or any settings. Also found out I had to disable the sysreq key. Thanks!

https://www.aliexpress.com/store/product/Custom-40-keys-IP65-fully-sealed-USB-PS2-RS232-interface-metal-keypad-keyboard-with-numbers-letters/1360552_32319513912.html?spm=2114.12010612.8148356.21.79aa519epmkg9b

bcherup, 2018-12-17:
You can also buy a 4 dollar keyboard from Microcenter (Inland brand) and
shave off the key inserts then epoxy it back together.

thewh1teagle, 2024-09-22:
Consider using [xremap](https://github.com/xremap/xremap). it works on x11/wayland and you can remap specific process.