guysoft/FullPageOS #266: Pi A can see all of Pi B's information

tomb456, 2019-01-18:
#### What were you doing?

Pi A has a user at a website log in. Pi B has that same website. When clicking log in on Pi B they have all of Pi A's credentials and they are logged into Pi A's account.

I'm not entirely sure if this is a FullPageOS issue, but I'm trying to cover all of my bases. 

#### What did you expect to happen?
The user's to be able to log in separately. 
#### What happened instead?
The user at Pi B had full website account access of Pi A. 
#### Was there an error message displayed? What did it say?
No error messages. 
#### Version of FullPageOS?

0.9.0

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-01-20:
Might be related to #81 ?
Did you change the hostnames?

tomb456, 2019-01-21:
Yeah,

The hostnames are changed and they have completely different IP addresses.

guysoft, 2019-01-22:
And did you run the command at: https://github.com/guysoft/FullPageOS/issues/81#issuecomment-262957567 ?

tomb456, 2019-01-24:
I did. I'm currently working on seeing if I can recreate the issue on a different network since the original one isn't available right now.

guysoft, 2019-01-26:
Ok, please update when possible, it might be a security issue.