guysoft/FullPageOS #267: Support of Pi 3 A+

DrNuss, 2019-02-13:
When do you think you will support the Raspberry Pi 3 A+ ?

#### What were you doing?
Displaying a website.

#### What did you expect to happen?
To see a website.

#### What happened instead?
on 0.9: colorful fullscreen picture before booting
on 0.10: after normal booting a blackscreen (sometimes with curser)

#### Was there an error message displayed? What did it say?
no

#### Version of FullPageOS?
0.9 and 0.10 (2018-12-20_2018-11-13)
Both with a Pi 3 A+

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-02-14:
Quite right, in the process of building a release candidate now that I finished releasing for OctoPi.

guysoft, 2019-03-10:
Support has been added to RC2, please test #270

guysoft, 2019-04-13:
Just tested RC3 both gpu and non-acceleration versions work

closed 2019-04-13 by guysoft