guysoft/FullPageOS #268: Help us test future FullPageOS release

tailorvj, 2019-02-15:
#### I am looking for people who are currently using FullPageOS to help testing the future release

I am currently working on a solution that allows you to easily rotate multiple URLs on a single Raspberry Pi and a cloud service that allows you to control the URL playlist on all of your FullPageOS installations remotely. 

Drop me a line on the [FullPageOS Facebook Page inbox](https://www.facebook.com/fullpageos), reply to this issue or contact me through [my Github profile](https://github.com/tailorvj)

#### How is this going to be better?

Our current rotation solution relies on iframes, which is very weird and causes all sorts of display problems. 

The future release will allow you to have a playlist file on the microSD card that will completely replace the URL in the browser. 

The future browser will be based on Electron. 

The remote control service will allow you to control a large amount of Raspberry Pi machines running FullPageOS and even group them, so you will be able to change the playlists on multiple machines at the same time. 

#### We need your feedback

Some initial releases are ready for you to test and provide direct feedback that will affect future development of FullPageOS. Join us!

guysoft, 2019-02-18:
@tailorvj Any build server at hand? Mine has moved to my parents house and is loosing VPN connection once in a while, I want to RC version 0.10.0 with A+ support but I keep loosing it

tailorvj, 2019-02-18:
@guysoft I don't have a build server at hand, but please let me know what kind of hardware/software/connection is required and I'll see if I can get someone to help out with this. 

FYI, I'm not doing automated builds yet. Just testing the rotation features on a Raspian Stretch. My current release is not even based on CustomPiOS, but I would eventually like it to be. At the moment, I'm simply building a customized Raspbian Stretch + xorg version (not lite, desktop without recommended software) and saving the img file from the microSD for testing purposes. 
A dockerhub solution is also being considered, but I may need assistance to make it happen. 

I would appreciate your input about the right/healthy build process for the official release. 

Please [join the test group mailing list](http://eepurl.com/ghHv4n) as well, so you would be in the loop about the process and feature development. I intend to provide an initial group of testers with the initial test system in the next few days, to start gathering feedback and prepare for a future official release. 

Cheers,
Tailor