guysoft/FullPageOS #269: Chromium-browser fails to start on auto-boot

fabilaj, 2019-03-08:
#### What were you doing?
Pi froze and need to be rebooted.  Power cycled and after booting I get a black screen in my vnc viewer.

#### What did you expect to happen?
Normally I get to see the web page loaded.

#### What happened instead?
Black Screen.

#### Was there an error message displayed? What did it say?
When I attempt to run "./run_onepageos" or "./start_chromium_browser" I get the following output:
$ ./start_chromium_browser
-bash: ./start_chromium_browser: No such file or directory
netops@netops-mn3:/$ /home/pi/scripts/./start_chromium_browser
/usr/lib/chromium-browser/chromium-browser: /lib/arm-linux-gnueabihf/libc.so.6: version `ELiBC_2.6' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /lib/arm-linux-gnueabihf/libc.so.6: version `LIBC_2.7' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX[3.4n19' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NSS_3.1".3' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NSSß3.10' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NWS_3.:' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NSS_3.9.6' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `JSS_3.9' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NSS_3.10.1' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnss3.so: version `NSS_3.11€SECMOD_ReleaseReadLock' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnssutil3.so: version `NSSUTIL_#&12' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libnssutil3.so: version `NSSUTIL_3.12.3NSS_VersionCneck' not found (required by /usr/lib/chromium-browser/chromium-browser)
/usr/lib/chromium-browser/chromium-browser: /usr/lib/arm-linux-gnueabihf/libsmime3.so: version `NSSß3.10' not found (required by /usr/lib/chromium-browser/chromium-browser)

When I attempt to run x11vnc manually as root I get:
08/03/2019 14:11:21 x11vnc version: 0.9.13 lastmod: 2011-08-10  pid: 12429
08/03/2019 14:11:21 XOpenDisplay("") failed.
08/03/2019 14:11:21 Trying again with XAUTHLOCALHOSTNAME=localhost ...
08/03/2019 14:11:21 
08/03/2019 14:11:21 *** XOpenDisplay failed. No -display or DISPLAY.
08/03/2019 14:11:21 *** Trying ":0" in 4 seconds.  Press Ctrl-C to abort.
08/03/2019 14:11:21 *** 1 2 3 4 
No protocol specified
08/03/2019 14:11:25 XOpenDisplay(":0") failed.
08/03/2019 14:11:25 Trying again with XAUTHLOCALHOSTNAME=localhost ...
No protocol specified
08/03/2019 14:11:25 XOpenDisplay(":0") failed.
08/03/2019 14:11:25 Trying again with unset XAUTHLOCALHOSTNAME ...
No protocol specified

#### Version of FullPageOS?

0.9.0

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-03-09:
Hey,
Did you try using the non-acceleration version?
Sounds like GPU issues

fabilaj, 2019-03-13:
Thanks for the reply, we ended up reflashing the card.  I'll look into your suggestion if this creeps up again.  Cheers.

closed 2019-03-13 by fabilaj