guysoft/FullPageOS #270: FullPageOS 0.10.0 RC2 status

guysoft, 2019-03-10:
FullPageOS 0.10.0 second release candidate.

Features:
* Chromium Version 65.0.3325.181 (Official Build) Built on Raspbian , running on Raspbian 9.4 (32-bit)
* New no-acceleration variant, basically the identical to main variant, but with GPU not enabled. The reaons is some users reported black screens on some RaspberryPis, and disaaling the GPU seems to solve it. Info at #235 #179 #128
* New Built on new Raspbian base image 2018-11-13
* Add option to Auto refresh added on log errors [WIP] thanks to @benyanke  ( https://github.com/guysoft/FullPageOS/commit/379204e7ffc88c1f62324e4eca34af9dd00b3442 )

Changes since RC1:

* Support for RaspberryPi 3 A +
* Updated to Raspbian base image 2018-11-13

Please test and report if there are any problems

## Download
http://unofficialpi.org/Distros/FullPageOS/nightly/2019-03-09_2018-11-13-fullpageos-stretch-lite-0.10.0.zip
md5: [``81334ae191627eee883198a154dd3587``](http://unofficialpi.org/Distros/FullPageOS/nightly/2019-03-09_2018-11-13-fullpageos-stretch-lite-0.10.0.zip.md5)

## no-acceleration variant

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2019-03-09_2018-11-13-fullpageos-no-acceleration-stretch-lite-0.10.0.zip
md5: [``4c76c42b2019c1a4df90db8ee2ad4771``](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2019-03-09_2018-11-13-fullpageos-no-acceleration-stretch-lite-0.10.0.zip.md5)

guysoft, 2019-03-10:
Known issue is the zero day bug in chromium, see: https://www.raspberrypi.org/forums/viewtopic.php?t=235291

We first need to see if a fix would even be released.

DrNuss, 2019-03-20:
I just tried "FullPageOS 0.10.0" on the RaspberryPi 3 A+.
It boots, gets the IP adress and then ther is a sort of blackscreen with a white short minus in the left upper corner. Sometimes this lasts for ever sometimes it lasts for like 10 minutes. After that it opens the browser and the website. The lights PWR and ACT shine all the time during this process.

With the non-accelerated version its the same but the blackscreen is without this little minus in the corner. After 7 minutes it opens the browser and the website (https://www.google.com).

With the normal Raspbian I can almost watch YouTube videos fluid on that Pi. So it's not a hardware problem...

guysoft, 2019-03-25:
Hey. Will note I am aboard so I won't be able to test yet.  Hope to get to around second week of April.

guysoft, 2019-04-11:
Note image released, with Chromium browser updated to version 72, likely because of zero day bug in chromium from https://github.com/guysoft/FullPageOS/issues/270#issuecomment-471282578 , expect RC2

guysoft, 2019-04-13:
Please use RC3 #273

closed 2019-04-13 by guysoft