guysoft/FullPageOS #271: Some uBlock origin update just broke this tool

bhartvigsen, 2019-03-14:
Hopefully this saves some trouble for somebody. Apparently uBlock origin (which is bundled with the Chromium on this tool for some reason) pushed out an update that broke a bunch of stuff everywhere. It brought down inventory tracking in our manufacturing facility until I disabled it. My fix has been:

rm -rf ~/.config/chromium/Default/Extensions/cjpalhdlnbpafiamejdnhcphjbkeiagm
sudo reboot

(the directory on the end might vary for you, I don't know)

guysoft, 2019-03-18:
Thanks for reporting. Which version of FullPageOS? And which version of Chromium?

bhartvigsen, 2019-03-18:
I'm not sure how to identify the version, it's a year or so old. Uname -a shows:
Linux fullpageos 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

I have this in my dpkg:

ii  chromium-browser                      56.0.2924.84-0ubuntu0.14.04.1.1000        armhf        Chromium web browser, open-source version of Chrome
ii  rpi-chromium-mods                     20170809                                  armhf        Raspberry Pi-specific mods to Chromium

bhartvigsen, 2019-03-18:
I get it.

The fix is to remove adblocker plugins from a kiosk distribution, have you done that? I can't imagine any reason an adblocker would need to be included in something like this.

closed 2019-03-18 by bhartvigsen

guysoft, 2019-03-28:
Its included because its part of the default shipped with raspbian.