guysoft/FullPageOS #272: use xdotool to press a key after a page has loaded

gbrakcounty, 2019-04-10:
#### What were you doing?
Trying to use xdotool to press enter at a login page that has the creds saved.

[Please be as specific as possible here]

#### What did you expect to happen?
i put this at the end of the script ~/scripts/start_chromium_browser so the full line looks this:
chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt) && sleep 60s && xdotool search --onlyvisible --class "chromium-browser" windowfocus && xdotool key Return
#### What happened instead?
since the creds are saved, pressing Enter manually will initiate login, but it doesnt seem to work using xdotool
#### Was there an error message displayed? What did it say?
No error message
#### Version of FullPageOS?
i dont have /etc/fullpageos_version, but i am definitely using fullpageos
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

gbrakcounty, 2019-04-11:
i want to update the issue:
so i edited the string at the end of the start_chromium_browser.sh script so it now looks like this:
chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt) & sleep 15s && xdotool key Return.  So when the pi boots, chromium loads the page, the creds are saved, Enter is press via xdotool, logs in, great.  Now the problem is that the script is being looped it seems and multiple instances of chromium are running because of it.  the script seems to be looping.  Forgive my noobness when it comes shell scripting im new to this.

guysoft, 2019-04-11:
The last command in the script means to be blocking. A blocking command means it hangs there running.

What you did is forked chromium using the ``&`` symbol, and then added a non-blocking command. Or well, a command that terminated after it finishes. You want chromium to be the last command. So just switch the order.
```
# fork and wait for chromium to load
sleep 15s && xdotool key Return  &

#run a blocking command that will not terminate
chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt) & sleep 15s
```
You might need more seconds, letting chromium load.

gbrakcounty, 2019-04-12:
> Attention: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.

Oh ok.  So should I use the code you provided below exactly how its shown and place it in ~/scripts/start_chromium_browser.sh?  15s is actually enough I clocked chromium’s startup from a cold boot at about 12 seconds from the time that the mouse cursor appears/desktop environment loads.

gbrakcounty, 2019-04-12:
I just tried your suggestion exactly how you typed it and the script still seems to be looping start new instances of chromium which eventually pins the memory.  Where is start_chromium_browser.sh being called from?  Maybe I can add a custom script for the keypress at startup independent from the chromium script?

guysoft, 2019-04-13:
This should work
```
# fork and wait for chromium to load
(sleep 15s ; xdotool key Return ) &

#run a blocking command that will not terminate
chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt) & sleep 15s
```

What happends if you comment it all out. like this?

```
# fork and wait for chromium to load
#(sleep 15s ; xdotool key Return ) &

#run a blocking command that will not terminate
chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt) & sleep 15s
```
It might be looping because of something else.

gbrakcounty, 2019-04-15:
Tried both sets of code you suggested, still loops.  What I’m thinking now is to leave the start_chromium_browser.sh alone and trigger a separate script to sleep press the keys at the same time that the start_chromium_browser.sh is launched, this way they coincide instead of being mixed together.  Where does the start_chromium_browser.sh get called from?  Maybe I can place my own script to launch from there so they run parallel to each other.

guysoft, 2019-04-17:
Ok, I haven't tested the fix I gave, and I can't see exactly what you are doing and how to set it up.

guysoft, 2019-06-23:
Closing if no reply

gbrakcounty, 2019-06-23:
Oh yeah I almost forgot. I found out how to open multiple pages. I have to add them in start_chromium_browser in ~/scripts