guysoft/FullPageOS #273: FullPageOS 0.10.0 RC3 status

guysoft, 2019-04-13:
FullPageOS 0.10.0 third release candidate.

Features:
* Chromium Version 72.0.3626.121 (Official Build) Built on Raspbian , running on Raspbian 9.8 (32-bit)
* New no-acceleration variant, basically the identical to main variant, but with GPU not enabled. The reaons is some users reported black screens on some RaspberryPis, and disaaling the GPU seems to solve it. Info at #235 #179 #128
* New Built on new Raspbian base image 2019-04-08
* Add option to Auto refresh added on log errors [WIP] thanks to @benyanke  ( https://github.com/guysoft/FullPageOS/commit/379204e7ffc88c1f62324e4eca34af9dd00b3442 )
* Support for RaspberryPi 3 A +

Changes since RC2:
* Updated to Raspbian base image 2019-04-08
* Updated chromium to version 72, which fixes zero day bug in chromium, see: https://www.raspberrypi.org/forums/viewtopic.php?t=235291

Please test and report if there are any problems

## Download
http://unofficialpi.org/Distros/FullPageOS/nightly/2019-04-12_2019-04-08-fullpageos-stretch-lite-0.10.0.zip
md5: [``383364675e973bbda92d0d1605740ec4``](http://unofficialpi.org/Distros/FullPageOS/nightly/2019-04-12_2019-04-08-fullpageos-stretch-lite-0.10.0.zip.md5)

## no-acceleration variant

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2019-04-12_2019-04-08-fullpageos-no-acceleration-stretch-lite-0.10.0.zip
md5: [``beda857dd6a33c8267a1e8121fac1a24``](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2019-04-12_2019-04-08-fullpageos-no-acceleration-stretch-lite-0.10.0.zip.md5)

guysoft, 2019-04-13:
I tested both images on A+, seems to work.

advorkina, 2019-05-03:
On b+ works well too. Thank you very much! 

Not sure if the question is related to this topic, sorry if not. 
But how can I enable auto refresh option if browser crashes? Should I make mine own build and uncomment the 2 lines that were mentioned here [379204e](https://github.com/guysoft/FullPageOS/commit/379204e7ffc88c1f62324e4eca34af9dd00b3442)? Or there is an easier approach? Thank you!

guysoft, 2019-05-23:
@advorkina You can just comment those in an existing installation.
You are always welcome to make your own build though.

guysoft, 2019-06-18:
Use RC4, likely this is finally the release since its really the same as the RC.
#281

closed 2019-06-18 by guysoft