guysoft/FullPageOS #274: X or GUI freezes

OliverGruenberg, 2019-05-13:
Hi there,
I have a strange behaviour here on some of our FullPageOS systems...
Some of them "freeze" or better: the user interface in chromium freezes all of a sudden.

I am not sure if this is a memory related issue. To be honest I don't really know where to look for huge log files or an identifiying entry. Also ncdu does not show any big files.

I've looked into ~/.config/chromium/chrome_debug.log and fount the latest entry to be:
[1255:1255:0509/084216.340613:ERROR:chrome_browser_main_extra_parts_x11.cc(62)] X IO error received (X server probably went away)

Any idea?

guysoft, 2019-05-13:
1. What model of raspberrypi?
2. What version and variant of FullPageOS?

OliverGruenberg, 2019-05-15:
Raspberry Pi 3B
Linux box 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
Raspbian GNU/Linux 8.0 (jessie)

How do I find out the FullPageOS variant?
It's an installation dated 2018

guysoft, 2019-05-30:
There are two variants for new FullPageOS. the default enables the GPU, there is also one where its disabled. Please try the one where the GPU is disabled.
I also recommend you upgrade.

OliverGruenberg, 2019-06-19:
found it: it was USB bus related... system "froze" on heavy network load together with power instability.

closed 2019-06-19 by OliverGruenberg