guysoft/FullPageOS #275: Optimize "refresh after error" (scripts/start_chromium_browser)

Moonbase59, 2019-05-14:
When using the "refresh after error" feature in scripts/start_chromium_browser, using the default log level and setting VLOG (verbosity) to 1 might be a bad idea: Chromium’s log file gets VERY noisy and tends to fill up the SD card space rather quickly.

So, I suggest changing line 11 of "start_chromium_browser" from
```bash
chromium-browser --enable-logging --v=1 --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt | sed -e "s/{serial}/${SERIAL}/g") &
```

to
```bash
chromium-browser --enable-logging --log-level=2 --v=0 --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --app=$(head -n 1 /boot/fullpageos.txt | sed -e "s/{serial}/${SERIAL}/g") &
```

whichs sets the log_level to 2 (ignore INFO and WARNING, show only ERROR and up) and verbosity to 0. This helps keeping the log file down to the minimum required and still shows all ERROR log entries.

Tested with FullPageOS 0.9.0 and 0.10, using Chromium 72.0.3626.121.

guysoft, 2019-05-23:
You are welcome to send a pull request :)

guysoft, 2019-08-19:
Fixed by #294

closed 2019-08-19 by guysoft